facebookarchive/Surround360 #3: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'

jasjuang, 2016-07-26:
I am on Ubuntu 14.04 LTS. I followed the exact same steps listed in the Readme file in surround360_render. When compiling the below error shows 

``````
[ 42%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/CvUtil.cpp.o
[ 46%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/JsonUtil.cpp.o
[ 50%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/StringUtil.cpp.o
Linking CXX static library lib/libLibVrCamera.a
[ 50%] Built target LibVrCamera
[ 53%] Building CXX object CMakeFiles/Raw2Rgb.dir/source/camera_isp/Raw2Rgb.cpp.o
Linking CXX executable bin/Raw2Rgb
/usr/bin/ld: CMakeFiles/Raw2Rgb.dir/source/camera_isp/Raw2Rgb.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [bin/Raw2Rgb] Error 1
make[1]: *** [CMakeFiles/Raw2Rgb.dir/all] Error 2
make: *** [all] Error 2
``````

This is tested in both release mode and debug mode

apparrapo, 2016-07-27:
Hi @jasjuang, can you try replacing line 31 in

https://github.com/facebook/Surround360/blob/master/surround360_render/CMakeLists.txt

with

SET(PLATFORM_SPECIFIC_LIBS "-lpthread")

and see if that solves the problem?

Thanks :)

jasjuang, 2016-07-27:
It doesn't. However I figured out if I edit line 21 to be

``````
SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread")
``````

It compiles fine for me.
 

closed 2016-07-27 by jasjuang

hrtain, 2018-11-13:
Tanks, it is a perfect idea for my question.

kamdoumloich, 2019-01-22:
Adding -pthread to C++ compiler flags worked for me

edisonlo, 2019-03-05:
> Adding -pthread to C++ compiler flags worked for me

this solved the problem!!!

SylviaZeng, 2019-03-12:
> Adding -pthread to C++ compiler flags worked for me

it works for me too！thank you!

ruiyuanlu, 2019-06-27:
> It doesn't. However I figured out if I edit line 21 to be
> 
> ```
> SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread")
> ```
> 
> It compiles fine for me.
> �

It works for me too, thank you!