facebookarchive/Surround360 #4: CMake should not be built in source

jasjuang, 2016-07-26:
In the instruction of surround360_render, it is showing building in source 

``````
cd /surround360/surround360_render
cmake -DCMAKE_BUILD_TYPE=Release
make
``````
which is a bad practice (http://stackoverflow.com/questions/1208681/with-cmake-how-would-you-disable-in-source-builds). In source build should be disabled and the instructions should look something like 

``````
cd /surround360/surround360_render
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release ..
make
``````

ybsheen, 2016-07-27:
Same issue shows in opencv build instructions too.

vensuka, 2016-07-30:
This also made OpenCV go off for me :O

russellkaye, 2016-08-22:
Same issues here.

s8i, 2016-08-29:
I am getting build issue as well
```
8i at shane in ~/Sites/Surround360/surround360_render/XCodeDebug on master [?]
$ cd /Users/8i/Sites/Surround360/surround360_render

8i at shane in ~/Sites/Surround360/surround360_render on master [?]
$ mkdir build

8i at shane in ~/Sites/Surround360/surround360_render on master [?]
$ cd build

8i at shane in ~/Sites/Surround360/surround360_render/build on master
$ cmake -DCMAKE_BUILD_TYPE=Release ..
-- The CXX compiler identification is AppleClang 7.3.0.7030031
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/8i/Sites/Surround360/surround360_render/build

8i at shane in ~/Sites/Surround360/surround360_render/build on master
$ make
Scanning dependencies of target LibJSON
[  2%] Building CXX object CMakeFiles/LibJSON.dir/source/thirdparty/supereasyjson/json.cpp.o
[  4%] Linking CXX static library lib/libLibJSON.a
[  4%] Built target LibJSON
Scanning dependencies of target LibVrCamera
[  6%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/SystemUtil.cpp.o
[  9%] Building CXX object CMakeFiles/LibVrCamera.dir/source/calibration/IntrinsicCalibration.cpp.o
[ 11%] Building CXX object CMakeFiles/LibVrCamera.dir/source/calibration/KeypointMatchers.cpp.o
In file included from /Users/8i/Sites/Surround360/surround360_render/source/calibration/KeypointMatchers.cpp:38:
/usr/local/include/glog/logging.h:483:9: warning: 'LOG' macro redefined [-Wmacro-redefined]
#define LOG(severity) COMPACT_GOOGLE_LOG_ ## severity.stream()
        ^
/usr/local/include/opencv2/stitching/detail/util.hpp:85:9: note: previous definition is here
#define LOG(msg) LOG_(1, msg)
        ^
1 warning generated.
[ 13%] Building CXX object CMakeFiles/LibVrCamera.dir/source/calibration/RingRectification.cpp.o
[ 16%] Building CXX object CMakeFiles/LibVrCamera.dir/source/optical_flow/NovelView.cpp.o
[ 18%] Building CXX object CMakeFiles/LibVrCamera.dir/source/optical_flow/OpticalFlowVisualization.cpp.o
[ 20%] Building CXX object CMakeFiles/LibVrCamera.dir/source/render/CameraMetadata.cpp.o
[ 23%] Building CXX object CMakeFiles/LibVrCamera.dir/source/render/ImageWarper.cpp.o
[ 25%] Building CXX object CMakeFiles/LibVrCamera.dir/source/render/PoleRemoval.cpp.o
[ 27%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/CvUtil.cpp.o
[ 30%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/JsonUtil.cpp.o
[ 32%] Building CXX object CMakeFiles/LibVrCamera.dir/source/util/StringUtil.cpp.o
[ 34%] Linking CXX static library lib/libLibVrCamera.a
[ 34%] Built target LibVrCamera
Scanning dependencies of target TestExtractAlphaChannel
[ 37%] Building CXX object CMakeFiles/TestExtractAlphaChannel.dir/source/test/TestExtractAlphaChannel.cpp.o
[ 39%] Linking CXX executable bin/TestExtractAlphaChannel
ld: library not found for -lgflags
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/TestExtractAlphaChannel] Error 1
make[1]: *** [CMakeFiles/TestExtractAlphaChannel.dir/all] Error 2
make: *** [all] Error 2

8i at shane in ~/Sites/Surround360/surround360_render/build on master
$
```

jingxw, 2016-09-14:
Hi s8i, I got the same build issue. Have you solved it? could you share the solution? Thanks!