facebookarchive/Surround360 #5: 80 character line length should also apply to comments

jasjuang, 2016-07-26:
In the contribution section, the 80 character line length is mentioned in the coding style guideline. 

I believe this restriction not only should be applied to the code but also to the comments. I am scanning through the code with Sublime in multi-column mode. It works great browsing the code but not so great when I want to read the comments, and I end up having to scroll left and right with my mouse in order to read them.

jasjuang, 2016-07-26:
Just realize some code does not obey this rule either... For example PixFlow.h

fbriggs, 2016-07-26:
There are some places where we have deliberately chosen to exceed 80 characters (e.g., the gflags argument lists at the top of most files in /surround360_render/source/test).

We also accidentally had a text editor set to 90 characters instead of 80, but fixing it was lower priority than other issues pre-release.

Your attention to detail is appreciated, and we will fix some of this in future updates.

BroadDong, 2016-07-28:
hello,could you give me a contact?I have some questions.

closed 2017-01-03 by fbriggs