facebookarchive/Surround360 #7: Cameras overheat?

naggie, 2016-07-26:
From the website:

>  and the industrial cameras can run for hours without overheating.

Does this imply they will overheat eventually? Or can this happily run 24/7? I guess adding a fan isn't a big deal.

Thanks for open sourcing.

fbriggs, 2016-07-26:
It is common for high end cameras to produce a lot of heat. However, these cameras are designed for industrial applications, and we have kept them on for several weeks without a problem. You will probably run out of space to store the recorded data long before anything overheats.

naggie, 2016-07-26:
Excellent, sounds like they have a stable steady state.

closed 2016-07-26 by naggie

fbriggs, 2016-07-26:
The cameras are also sandwiched between two metal plates, which acts as a heat sink.