facebookarchive/Surround360 #10: Solidworks CAD data files?

justinclift, 2016-07-27:
Out of curiosity, would it be possible to provide the source [Solidworks 2016](https://github.com/facebook/Surround360/blob/master/surround360_design/3d_models/STEP_files/160523_3.STEP#L9-L10) CAD files as well?

STEP is good for CAD data interchange, but for the people who have Solidworks 2016 themselves the native models would be more useful. :smile:

bkcabral, 2016-07-27:
Thanks for the feedback.  We'll look into it and if we can support that.  We're mainly worried about file size and maintaining multiple versions of the same design - least they get out of sync.

closed 2017-01-03 by fbriggs

justinclift, 2017-01-03:
So, not going to happen?

fbriggs, 2017-01-03:
Just closing the task for book-keeping since it hasn't been updated in ~5 months.
Is this something you really need/will actually use now? If so I can poke the relevant people again.

justinclift, 2017-01-03:
Nah, it was a general thought only.  No specific use case personally. :smile: