facebookarchive/Surround360 #11: Less cameras

EnriqueMVega, 2016-07-27:
What would be the minimum number of cameras to use? I love the design but there are so many cameras and it gets expensive. I wonder if I can do my own design with less cameras and maybe increase the fov with different lenses

Are you going to have this for windows?

fbriggs, 2016-07-27:
One parameter of our rendering system is the interpupilary distance (IPD) that is simulated. The population median IPD is near 6.4cm, so we use that value. Depending on how many cameras you have, the radius of the rig, and the horizontal FOV of the lenses, you may be able to simulate more or less IPD. In general more cameras gives more IPD, more FOV gives more IPD, and a larger rig radius gives more IPD. However, increasing FOV decreases pixel density, and increasing the rig radius makes optical flow more difficult. The formula is (in degrees, not radians):

i = r * sin( FOV/2 - 360 / n)
where
i = 1/2 the IPD that we can simulate
r = the radius of the camera array (technically, the distance from the center of the rig to the exit pupil of the lenses)
FOV = horizontal FOV of the side cameras
n = number of cameras

As long as you have i > 3.2cm, you can simulate full 6.4cm IPD stereo.

Please note also that your software does not fully support 180 degree fisheye lenses for the side cameras, although the modifications to get full support are relatively minor. For example, we don't currently support intrinsic parameters to correct for lens distortion in fisheye lenses, and we don't have a way to do joint stereo rectification. If you are interested in knowing more about what changes need to be made, please follow up.

akc2267, 2016-10-04:
Hi fbriggs! Can you please post about the changes necessary to make 180degree fisheye lenses work for the side cameras?

fbriggs, 2016-10-05:
See answer here: https://github.com/facebook/Surround360/issues/73

Other relevant discussions:
https://github.com/facebook/Surround360/issues/94
https://github.com/facebook/Surround360/issues/97
https://github.com/facebook/Surround360/issues/109

closed 2016-10-13 by fbriggs