facebookarchive/Surround360 #15: Increase FOV

EnriqueMVega, 2016-07-28:
I would like to increse the default 90 deg FOV in the camaras for fisheye or full frame fisheye. What changes do I have to do in the code to adjust/remove distorsion? I hope to use less cameras increasing the fov. Thanks

EnriqueMVega, 2016-07-28:
Related to the question, What is the max fov that we can use and still get good 3d?

fbriggs, 2016-07-28:
Our system already has partial support for using fisheye lenses on the side cameras, but it is incomplete and there may be issues. In order to use fisheye lenses on side cameras, one specifies the is_fisheye property and other related properties in in the json file that describes the configuration of of lenses. I have attached an example of such a file with this message.

[8pi180.json.zip](https://github.com/facebook/Surround360/files/388953/8pi180.json.zip)

Currently, we do not support intrinsic calibration for fisheye lenses, which means it is assumed in the rendering code that they are perfect f-theta lenses. Deviation from this may introduce distortion. More work would be required to correct this distortion. When running the rendering software with fisheye lenses on the side, you should pass NONE for the --src_intrinsic_param_file.

We also do not rectification with fisheye lenses on the side, which may be important to produce comfortable stereo results. This could be solved with perfect bundle adjustment. The key issue is that early in the rendering pipeline, the side camera images are projected from their original form to equirectangular projection. With the regular pinhole cameras we use on the side now, we specify a parameter --ring_rectify_file, which gives a transform of the images such that they are all mutually stereo-rectified in the resulting equirectangular projection. You can instead pass NONE for --ring_rectify_file (we provide code to estimate the rectification transforms, but it is not applicable to fisheye lenses).

One other known bug with using fisheye lenses for side cameras is that the projection to equirectangular is only correct if the FOV specified in the json file is exactly 180 degrees. Different values from 180 will result in some vertical stretching. This is relatively simple to fix, and we might do it in the not too distance future.

fbriggs, 2016-07-28:
There is no max FOV to still get good 3D. The relevant concern is- if you use a fisheye lens, the angular pixel density is lower compared with a lower FOV lens on the same resolution sensor. This can contribute to a "flattening" effect in the 3D.

Another problem with fisheye lenses is that they tend to be blurry and have issues with chromatic abberation at the edges. If you are using a small number of fisheye lenses, you will mostly be sampling pixels from the edges, which results in lower quality output.

closed 2016-07-28 by fbriggs

jilternj, 2016-08-17:
Hi, there,
    I just checked the code fix for non-180 FOV, however, it seems not work for FOV larger than 180. And another issue here I 'd like to know, is that  for fish eye lens there is no perspective transform and ring rectify, which means the optic flow to stich the side images is enough ?
    Another concern is that if there is any way to calibrate or model fish eye lens ? 
    your answer will be greatly appreciated. Thanks.

fbriggs, 2016-08-17:
It is possible that there is a bug for FOV > 180, or it may be an issue with how you set it up. If you post the data you are using + all commands you have run, I can take a look and see if there is any issue with configuration.

Regarding intrinsic calibration and rectification for side cameras using fisheye lenses, this is not currently implemented. Optical flow probably isn't enough on its own to produce usable results. Support for side-camera fisheye lenses is still a work in progress.

jilternj, 2016-08-18:
[image]

I used a picture as attached, which has FOV 220,max radius 1300 ,just call function testSideFisheyeProjection, and the result seems incorrect as the vertical part with broken.  Maybe I 
shall crop the vertical side to 180 degree to test ?

And for the fish eye support , would you like to share some detail math model you will use to support, I feel quite confusing after search a lot on how to do fish eye projection, I mean , not sure how to do homography  for fish eye lens. Or if any schedule when will you release your support. Thanks.

fbriggs, 2016-08-18:
I still need to see any config files you changed, and whatever commands or code you are running. If you provide these, I can help you debug it.

Regarding a mathematical model, look at "f-theta" lens projection models. That is the model used in our code.

jilternj, 2016-08-19:
I changed the file TestImageWarpper with 
 camModel.usablePixelsRadius = 1300;
 camModel.imageCenterX = srcImage.cols/2;
  camModel.imageCenterY = srcImage.rows/2; 
 camModel.fovHorizontal =220;
 camModel.aspectRatioWH = 1;

And generate TestImageWapper.exe and get the broken result.

fbriggs, 2016-08-19:
I tried with your image. I had to set a few more parameters in camModel to get it to work. Here's what worked for me:

`void testSideFisheyeProjection() {
  const string srcPath = FLAGS_test_dir + "/cam0.png";
  Mat srcImage = imreadExceptionOnFail(srcPath, -1);
  CameraMetadata camModel;
  camModel.isFisheye = true;
  camModel.usablePixelsRadius = srcImage.cols/2;
  camModel.imageCenterX = srcImage.cols/2;
  camModel.imageCenterY = srcImage.rows/2;
  camModel.aspectRatioWH = 1.0;
  camModel.fovHorizontal = 220;
  camModel.fisheyeFovDegrees = 220;
  camModel.fisheyeFovDegreesCrop = 220;
  camModel.fisheyeRotationDegrees = 0;
  Mat eqrImage = sideFisheyeToSpherical(srcImage, camModel, srcImage.cols, srcImage.rows);
  imwriteExceptionOnFail(FLAGS_test_dir + "/eqr.png", eqrImage);
}`

Result:
[image]

jilternj, 2016-08-21:
It worked. Thanks!
And hope to get to know your progress on side fish eye stiching in near future.