facebookarchive/Surround360 #17: frames for testing

EnriqueMVega, 2016-07-28:
So far the project looks great. Thanks a lot
Before we purchase the hardware I would like to test the render part. Are you able to share a set of frames (5 secs to 15 sec) from each camera so we can use it for testing?  I know it is a lot to ask but will really help us to understand what the software is doing.

VRabbitHole, 2016-07-28:
Agree with OP .... are there any RAW camera files we can take through the rendering software while we're waiting for the hardware?

vensuka, 2016-07-29:
Yep I think this will be great :)

sekibara, 2016-07-29:
it will be really helpful!  if worry about size, please post sample raw photos instead of video at first. thanks

goldonkey, 2016-08-02:
Thanks for sharing  your intellegent work！Captured sample  raw data file is  a very important part for everyone to try the OP code.

PaulBergmann, 2016-08-03:
Thank you for releasing this code to the public! I also think some test data would be highly useful to get a better understanding of the whole pipeline!

fbriggs, 2016-08-03:
here are 2 frames from the palace dataset: https://fburl.com/399375715
we will provide a larger dataset ASAP

VRabbitHole, 2016-08-03:
Just checked out the links on both threads and got this message:

Sorry, this page isn't available
The link you followed may be broken, or the page may have been removed.

fbriggs, 2016-08-03:
sorry, bad link. try this one instead:
https://s3-us-west-2.amazonaws.com/surround360/sample/sample_dataset.zip

closed 2016-08-03 by fbriggs