facebookarchive/Surround360 #23: Error installing gflags (method 1 - Linux only):

VRabbitHole, 2016-07-30:
vrlf@vrlf:/media/sandshare$ sudo apt-get install libgflags2 libgflags-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libgflags2 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libgflags2v5:i386 libgflags2v5

E: Package 'libgflags2' has no installation candidate

I'm installing on Ubuntu 16,04 LTS .... on BTRFS share.

aparrapo, 2016-07-30:
We have not tested the installation on Ubuntu 16.04 LTS yet, so I'm guessing the gflags package has changed.

I assume `libgflags2v5` would work as well.

Else, check the possible options via

`apt-cache search gflags`

Let me know if that doesn't work and I'll be more than happy to help :)

VRabbitHole, 2016-07-30:
I tried libgflags2v5 but Installer couldn't find package.

Sent from my iPad

> On Jul 30, 2016, at 11:14 AM, aparrapo <notifications@github.com> wrote:
> 
> libgflags2v5

VRabbitHole, 2016-07-31:
Issue fixed -thanks.

aparrapo, 2016-07-31:
How did you fix it? Did you find another available package?

VRabbitHole, 2016-07-31:
Updated everything I could .... went through the whole install process again .... inserted libgflags2v5 this time around and install worked just fine ... I did do a reboot of the PC before starting second install ... not sure what exactly fixed the problem but all is good now.

Might be useful to compile a fresh install version for Ubuntu 16.04 LTS ... it's been out a while now and is pretty stable and has latest version of BTRFS and ZFS included plus all the other latest packages. I know you guys recommend the Areca 8 x bay RAID and hardware RAID card for the RAW storage device but SDS (Software Defined Storage) with SSDs and nvme is the way to go for fast flash storage ... lighter, faster, cheaper and with Brtfs and ZFS better data integrity ... I've set up a 16 x SSD pool in half the space/weight required by the Areca 8 x bay chassis .... just a thought. I used eight Areca RAIDS on a WB movie a couple of years ago ... very reliable for sure but heavy as hell ... try lugging eight of those around the US/Canada and watch your biceps increase in size :-)

aparrapo, 2016-08-01:
Thanks for the advice! We are looking at other storage systems as well. And yes, we've been getting some workouts with the Areca boxes :)

VRabbitHole, 2016-08-01:
Be very interested in your thoughts on SDS storage architectures with SSDs and NVMe ... it's evolving rapidly ... please keep us in the loop on developments .... I'm presenting a short talk at the Flash Memory Summit next month in Santa Clara on 'Flash Enhanced Storage for 4K and VR Workflows' ... plus, lots of interest in VR in LA at the moment.

closed 2016-08-01 by fbriggs