facebookarchive/Surround360 #24: Error installing Gooey (method 2 - Linux only):

VRabbitHole, 2016-07-30:
vrlf@vrlf:~/ffmpeg_sources/ffmpeg$ sudo apt-get install python-wxgtk2.8
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package python-wxgtk2.8 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-wxgtk2.8' has no installation candidate

aparrapo, 2016-07-30:
Apparently it's a known problem in Ubuntu 16.04:

http://askubuntu.com/questions/789302/install-python-wxgtk2-8-on-ubuntu-16-04

It seems that you need to add the required repo to the package list.

VRabbitHole, 2016-07-31:
Issue fixed - thanks.

aparrapo, 2016-07-31:
Close#24

closed 2016-07-31 by fbriggs