facebookarchive/Surround360 #28: Low output quality

mimiaofandisp, 2016-08-01:
Hi, I generated a stereo panorama with the default setting, but the output panorama has very low quality. 

My inputs are 14 4000*3000 gopro images, I do not have top and bottom images. After the spherical projection, I checked the warped result, the quality is good with 2048*2048 resolution. I am not sure in which step the quality dropped so much. Here is a cropped region from the whole pano.
[image]

fbriggs, 2016-08-01:
In order to use the rendering code with GoPro images, you will need to specify a custom .json file for the rig which defines the number of cameras and their FOV. You will also need to go through the process of intrinsic calibration, as described in CALIBRATION.md.

Can you verify that you did those steps? If so, please post your .json file, the intrinsic .yml file, and any other data you used.

If you are using the GUI to in run_all.py, it is also helpful if you take a screenshot of the configuration screen, so we can see all o fthe options you are specifying. Alternatively, when run in GUI mode, run_all.py prints all of its input arguments in the first block of its logs.

mimiaofandisp, 2016-08-01:
Hi,

Thanks for the reply. I know I need to do the calibration myself with gopro setting. But currenetly I used your default intrinsic file to do the calibration as I suppose the calibration is only used for vertical disparity removal and prealignment. At this moment, I only want to see the output panorama format and image quality (not stitching quality), so I do not really care about the accuracy of the calibration. But according to what you said, do you mean the calibration will also affect the stereo rendering quality (image quality instead of stitching quality)?

mimiaofandisp, 2016-08-01:
BTW, I modified the code a little bit. 

in the testrenderstereopano file
LOG(INFO) << "Creating stereo equirectangular image";
Mat stereoEquirect = stackVertical(vector<Mat>({sphericalImageL, sphericalImageR}));
imwriteExceptionOnFail(FLAGS_output_equirect_path, stereoEquirect);

the imwrite did not specify the output file extension, so I added **+ "/test.png"** into the code. But I do not know what extension you really want for the stereo output.

mimiaofandisp, 2016-08-01:
ok, problem solved, need to change eqr_width and eqr_height

closed 2016-08-01 by fbriggs