facebookarchive/Surround360 #30: Creating a panorama from the sample data

PaulBergmann, 2016-08-04:
Hello! 

Thanks a lot for providing some sample data to try out by ourselves! 

I was trying to stitch a panorama from these images using "TestRenderStereoPanorama".
So far, I only want to stitch without the top/bot fisheyes (therefore only using the side cameras).

My call of the program then looks like this:

./TestRenderStereoPanorama 
-src_intrinsic_param_file ../res/config/sunex_intrinsic.xml 
-rig_json_file ../res/config/17cmosis_default.json 
-imgs_dir ../sample_dataset/vid/000000/isp_out 
-output_data_dir ../sample_dataset/vid/000000/output_data
-eqr_width 252

Then the program yields the following error (opencv cannot write the specified file extension):

OpenCV Error: Unspecified error (could not find a writer for the specified extension) in imwrite_, file /opencv-3.1.0/modules/imgcodecs/src/loadsave.cpp, line 459
E0804 13:04:31.529429  8554 SystemUtil.cpp:50] Terminated with exception: /opencv-3.1.0/modules/imgcodecs/src/loadsave.cpp:459: error: (-2) could not find a writer for the specified extension in function imwrite_

Regarding this, I have some questions:
1) When using the standard options, the error occurs that the eqr_width must be divisible by the number of cameras used. The default value is 256, and by default no top cameras are used, leaving 14 
cameras where 256 is not divisible by 14. Am I correct, setting this value manually to for example 252 which is divisible by 14 then? 

2) Is the general approach im following to create the panorama from the isp processed images correct? or am I missing something? This information would help 
me to figure out if something with my opencv is wrong or I just made a basic mistake using the stitching tools. 

Thank you very much!

closed 2016-08-04 by PaulBergmann

Ahmad-Masood, 2016-08-10:
Can you please share your final command which you used. Actually I'm getting following error:
`
Terminated with VrCamException: failed to write image: /Users/AMS/Surround360/sample_dataset/vid/000000/output_data/flow_images/overlap_0_L.png
E0810 10:38:46.707176 3211264 SystemUtil.cpp:48] Terminated with VrCamException: failed to write image: /Users/AMS/Surround360/sample_dataset/vid/000000/output_data/flow_images/overlap_1_L.png
`

PaulBergmann, 2016-08-10:
Did you make sure that the flow_images folder exists in your ouput_data folder? Opencv will not create it for you and throw this error if the folder does not exist yet

PaulBergmann, 2016-08-10:
I basically gave up trying to use ./TestRenderStereoPanorama  directly, the python gui works just fine

Ahmad-Masood, 2016-08-10:
Paul thanks for quick reply. 

Actually now i'm stuck with the error which you was facing

`
E0810 10:48:58.151003 1889587200 SystemUtil.cpp:50] Terminated with exception: /Users/AMS/Surround360/opencv/modules/imgcodecs/src/loadsave.cpp:462: error: (-2) could not find a writer for the specified extension in function imwrite_
E0810 10:48:58.156620 1889587200 SystemUtil.cpp:37] 0   TestRenderStereoPanorama            0x000000010c1110d6 _ZN11surround3604util15printStacktraceEv + 54
E0810 10:48:58.156674 1889587200 SystemUtil.cpp:37] 1   TestRenderStereoPanorama            0x000000010c11134c _ZN11surround3604util16terminateHandlerEv + 412
E0810 10:48:58.156692 1889587200 SystemUtil.cpp:37] 2   libc++abi.dylib                     0x00007fff8c13619e _ZSt11__terminatePFvvE + 8
E0810 10:48:58.156705 1889587200 SystemUtil.cpp:37] 3   libc++abi.dylib                     0x00007fff8c135c12 _ZN10__cxxabiv1L22exception_cleanup_funcE19_Unwind_Reason_CodeP17_Unwind_Exception + 0
E0810 10:48:58.156720 1889587200 SystemUtil.cpp:37] 4   libopencv_core.3.1.dylib            0x000000010d0bdf36 _ZN2cv5errorERKNS_9ExceptionE + 854
E0810 10:48:58.156736 1889587200 SystemUtil.cpp:37] 5   libopencv_core.3.1.dylib            0x000000010d0be354 _ZN2cv5errorEiRKNS_6StringEPKcS4_i + 212
E0810 10:48:58.156750 1889587200 SystemUtil.cpp:37] 6   libopencv_imgcodecs.3.1.dylib       0x000000010c89cfbe _ZN2cvL8imwrite_ERKNS_6StringERKNS_3MatERKNSt3__16vectorIiNS6_9allocatorIiEEEEb + 446
E0810 10:48:58.156764 1889587200 SystemUtil.cpp:37] 7   libopencv_imgcodecs.3.1.dylib       0x000000010c89cd0b _ZN2cv7imwriteERKNS_6StringERKNS_11_InputArrayERKNSt3__16vectorIiNS6_9allocatorIiEEEE + 203
E0810 10:48:58.156777 1889587200 SystemUtil.cpp:37] 8   TestRenderStereoPanorama            0x000000010c128d1d _ZN11surround3604util22imwriteExceptionOnFailERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEERKN2cv3MatERKNS1_6vectorIiNS5_IiEEEE + 157
E0810 10:48:58.156793 1889587200 SystemUtil.cpp:37] 9   TestRenderStereoPanorama            0x000000010c102dfc _Z20renderStereoPanoramav + 20044
E0810 10:48:58.156810 1889587200 SystemUtil.cpp:37] 10  TestRenderStereoPanorama            0x000000010c10620e main + 14
E0810 10:48:58.156824 1889587200 SystemUtil.cpp:37] 11  libdyld.dylib                       0x00007fff8d0f85ad start + 1
E0810 10:48:58.156841 1889587200 SystemUtil.cpp:37] 12  ???                                 0x0000000000000009 0x0 + 9`

PaulBergmann, 2016-08-10:
Yes thats the error I got as well. Switching to the python script solved this problem for me, you can either use run_all.py and only execute the render step in the gui or run batch_process_video.py in the terminal, both things work for me.

Ahmad-Masood, 2016-08-10:
Paul thanks again. As you suggested i used batch_process_video.py script with following parameters.

python batch_process_video.py 
--root_dir=/Users/AMS/Surround360/sample_dataset 
--surround360_render_dir=/Users/AMS/Surround360/Surround360/surround360_render/ 
--start_frame=00000 
--end_frame=00001 
--quality=preview 
--rig_json_file=../surround360_render/res/config/17cmosis_default.json 
--src_intrinsic_param_file=../surround360_render/res/config/sunex_intrinsic.xml --flow_alg=pixflow_ultra

And Im getting following output (It looks like quite a low quality thing). Is this what i Expect ? 
[image]

PaulBergmann, 2016-08-10:
Have you tried increasing the quality to 6k instead of using -quality=preview?

Ahmad-Masood, 2016-08-10:
It outputs a blank file.

Ahmad-Masood, 2016-08-10:
[image]

fbriggs, 2016-08-10:
quality=preview is not intended to give results that are viewable in VR. the purpose of that mode is to render as fast as possible, e.g., if we have a long video and want to see which parts of it to render at higher quality. with quality=preview lots features are disabled.

Ahmad-Masood, 2016-09-01:
I was able to solve it by changing quality parameter. Thanks for helping me out. And I'm sorry for responding late.