facebookarchive/Surround360 #32: Parameter camera_ring_radius

PaulBergmann, 2016-08-05:
Hello! 

Could you explain to me what the parameter "camera_ring_radius" is used for in the stitching 
and how I have to set it to get proper results when stitching images (not necessarily images from the facebook surround camera but any images?). 

I only found this parameter description:

DEFINE_double(camera_ring_radius,         19.0,           "used for computing shift for zero parallax as a function of IPD.");

Why is it that if I set this value too small, the projected images are not properly merged together but a space is left between them in the overall panorama?

Thank you for your help!!

fbriggs, 2016-08-05:
The value you should use here is the distance from the center of the rig to the tip of a lens, in centimeters.

The reason you may see black strips is that there is not enough data to reconstruct the parts of the lightfield that are required. This can happen if you have changed the number of cameras, the FOV of the lenses, the rig radius, or the IPD to be rendered. There is a formula that describes the maximum IPD that a particular configuration can simulate.

i = r * sin(FOV/2 - 360/n)       (degrees, not radians)
where
r = radius of the rig
FOV = horizontal Fov of the cameras **after correcting for barrel distortion**
n = number of cameras
i = IPD/2

For full stereo, we render with IPD=6.4cm, therefore whatever configuration of cameras you pick, you must get at least i = 3.2cm out of this formula, or you will get those black stripes.

closed 2016-08-05 by fbriggs

initialneil, 2016-08-08:
Hi @fbriggs , I have a question for this equation you gave:
```
i = r * sin(FOV/2 - 360/n) (degrees, not radians)
```
because in extreme case when `FOV/2 == 360/n`, the `i` is **0** here, which sounds like you cannot make stereo  images out of it.
But if you give left half of FOV to the right eye and vice versa for the left eye, you **can** generate stereo images. Are they proper stereo images? And what should be the IPD for it?