facebookarchive/Surround360 #34: Rendering panorama with own sample data

PaulBergmann, 2016-08-05:
Hello! 

I managed to render the panorama with the provided sample data and now want to try to stitch my own 
images together. Therefore I adapted the configuration files (intrinsics and json file), I also created a new .yml file for the rectification as described in your configuration document. 

I have a problem with the stereo panorama creation after the optical flow step. The picture pairs for optical flow are generated nicely but the resulting images to be horizontally stacked as a panorama are not 100% filled with image data, so there is a black bar on the right where there is no data (this seems like some projection problem to me).

Are there some other parameters (maybe in the python script) that I have to change to get the stitching working with my own data? Which parameters should be generally altered?

Thank you!

PaulBergmann, 2016-08-05:
Especially im curious about if the software also works if another number of cameras is used. Im only using 14 cameras and it seems when I decrease the number of cameras further, the unfilled gaps between the images become larger although a full 360 view should still be creatable.

fbriggs, 2016-08-05:
In order to help, I will need some more information. Can you please post:
- the images you used
- the output you got
- any commands you executed
- any config files you changed

thanks.

closed 2016-08-09 by fbriggs

liu666666, 2016-09-05:
hello,i want to render panorama with own sample data,too. my images are from only 3  cameras which record the left, middle,right direction of a playground,  i want get the  panorama video of 180 rather than 360.Could you please tell me how to implement it with your nice code? and could you tell me what should i  write in my config files ? what about r and other parameters ?

fbriggs, 2016-09-08:
Hi liu666666, the Surround 360 rendering code is only designed to render full 360 degree panoramas with a ring of cameras. It doesn't support generating partial panoramas.