facebookarchive/Surround360 #38: cameranames.txt file not found

VRabbitHole, 2016-08-07:
Did a completely fresh install of Ubuntu 14.04 LTS and only installed  apps necessary for the 360render engine ... want to keep the system clean as possible - not even upgrading to 16.04 LTS.

Fired up the run_all.py GUI and keep getting this error message:

-------- Sun Aug 07 2016 15:45:42 PDT -------

python /Surround360/surround360_render/scripts/run_all.py --rectify_file /media/fbvr4/fbvol1/dest5/rectify.yml --src_intrinsic_param_file /Surround360/surround360_render/res/config/sunex_intrinsic.xml --data_dir /media/fbvr4/fbvol1/sample_dataset/vid/000000/raw --start_frame 0 --frame_count 0 --cubemap_format video --steps all --rig_json_file /Surround360/surround360_render/res/config/17cmosis_default.json --dest_dir /media/fbvr4/fbvol1/dest5 --pole_masks_dir /Surround360/surround360_render/res/pole_masks --quality 6k --cubemap_face_resolution 0 --cam_to_isp_config_file /Surround360/surround360_render/res/config/isp/cam_to_isp_config.json --flow_alg pixflow_low

** UNPACK ** [Step 1 of 6]

E0807 15:45:42.032326  4966 SystemUtil.cpp:48] Terminated with VrCamException: file read failed:/media/fbvr4/fbvol1/sample_dataset/vid/000000/raw/cameranames.txt
Aborted (core dumped)

UNPACK runtime: 0:00:00.129529

Can not find 'cameranames.tx' file anywhere on my system ... was it included in the download or do we need to generate it?

Also, in the Render gui app the top left argument says 'Data Directory' and underneath 'directory containing .bin files' ... exactly which directory do we need to point to in the Palace sample_download .zip you provided? ...  I can only see .bmp and .png image files in the 'vid' folder. Tried both.

VRabbitHole, 2016-08-08:
Issue Solved

fbriggs, 2016-08-08:
cameranames.txt is required to run the 'unpack' step. The sample dataset we provided has already had the unpack step run (and the isp step as well). We are not providing binary files suitable for testing the unpack at this time, but we probably will in the future.

closed 2016-08-08 by fbriggs