facebookarchive/Surround360 #39: Average run times?

VRabbitHole, 2016-08-08:
Be interested to know what kind of speeds folks are getting on their render times for the Palace Test dataset?

I've run the process twice and am getting around 15 minutes on dual proc Intel Xeon CPU E5-2697 v2 @ 2.70GHz PC with 64GB RAM:

Runtime #1:
UNPACK runtime: 0:00:00.130109
ARRANGE runtime: 0:00:00.033033
total frames: 3
ISP runtime: 0:00:23.239405
RECTIFY runtime: 0:14:07.659499
RENDER runtime: 0:00:44.247832
FFMPEG runtime: 0:00:04.559142
TOTAL runtime: 0:15:19.870323

Runtime #2:
UNPACK runtime: 0:00:00.157266
ARRANGE runtime: 0:00:00.033929
total frames: 3
ISP runtime: 0:00:23.814318
RECTIFY runtime: 0:13:39.033048
RENDER runtime: 0:00:44.261449
FFMPEG runtime: 0:00:04.437203
TOTAL runtime: 0:14:51.738422
 
Am about to try same thing on a i7 PC with 32GB of RAM but would like to know if others are getting much faster RECTIFY times - all my datasets are on NVMe drives so I don't think I'm I/O bound and the files are relatively small anyway.

VRabbitHole, 2016-08-08:
Just ran the render process on single Intel Core i7-6700K CPU @ 4.00GHz with 8 Cores and 32 GB RAM and got an average render time of just under 6 minutes .... so it would seem that Clock Speed trumps Cores by a big margin ... both machines had a Titan X install so GPU wasn't a factor and as Brian said the code at the moment is CPU optimized anyway.

UNPACK runtime: 0:00:00.098312
ARRANGE runtime: 0:00:00.016530
total frames: 3
ISP runtime: 0:00:26.666419
RECTIFY runtime: 0:05:02.159184
RENDER runtime: 0:00:25.246620
FFMPEG runtime: 0:00:00.001236
TOTAL runtime: 0:05:54.210694

Anybody testing on a faster i7 with more than 8 Cores?

Have the FB guys run the 2 x frame Palace dataset through your machines? ... what kind of benchmark speeds are you getting? Any optimizations we need to make in order to decrease render times?

Would of course be great to have more frames to play with - see if render times are directly correlated to the number of frames in the dataset? If the 2 x frame test at 6 minutes render time directly scales, then we're looking at somewhere around 90 minutes per second of footage at 30fps ... DITs will have hard time explaining that to post-supervisors impatiently waiting for their dailies :-(

VRabbitHole, 2016-08-08:
Overclocked the i7 to 4.4GHz and that brought down the TOTAL runtime: 0:05:26.032436.

I was watching the Ubuntu System Monitor during the runtime and it seemed that only 2 of the 8 Cores were being stressed during the RECTIFICATION phase - the other 6 were idling along at under 2% utilization. RAM usage never went above 2GB.

Are there any tweaks to optimize multi-threading during RECTIFICATION?

fbriggs, 2016-08-08:
make sure you compile with:
cmake -DCMAKE_BUILD_TYPE=Release
make

and not:
cmake .

or
cmake -DCMAKE_BUILD_TYPE=Debug

this makes a huge difference

fbriggs, 2016-08-08:
Keep in mind rectification needs to be run only once per camera, then you can re-use the rectify.yml file for different videos with the same camra (assuming it is run on a scene with sufficient features). Even if you run rectification once per video (which is not a crazy idea), it only happens for one frame, then it is applied to all subsequent frames.

VRabbitHole, 2016-08-08:
For sure .... always do the cmake -DCMAKE_BUILD_TYPE=Release option.  Also did a completely fresh install of Ubuntu 14.04 LTS and only installed libs for the 360 render app.

What kind of Runtimes are you seeing with the 2 x frame Palace dataset?

VRabbitHole, 2016-08-08:
mmhhmm ... what exactly do you mean by ... "Even if you run rectification once per video (which is not a crazy idea), it only happens for one frame, then it is applied to all subsequent frames."

If I'm understanding you correctly are you saying that once we've done the first 1 x frame rectification (and say it takes 4 minutes) we can then apply that same rectify.yml code to all subsequent frames very quickly ... so for example, if you have a five minute take, the remaining 9000 frames will be rectified instantaneously?

If so, presumably, the TOTAL runtime of the process will then only depend on the RENDER part of the process? ... which I'm assuming is related to the total number of frames in the take? ... ie, a ten minute take will take twice as long as five minute take to render?

We could definitely do with some more frames to test that hypotheses if possible. ... the more the merrier. 

Just so you know, the Grey Point cameras are on back-order until the end of October so we won't have our rigs for a few months yet.

fbriggs, 2016-08-08:
Yes, rectification is only computed against 1 frame, and then applied to all frames in a video with negligible impact on runtime. The total runtime to render a video is dominated by time in the 'isp' and 'render' steps.

VRabbitHole, 2016-08-08:
Thanks for the clarification .... much appreciated ... some of us are still learning FB terminology ... quite different to what we've been used to for last couple of years in VRland ... very impressed with where you guys are going and how much engineering effort and smarts has already been put into the project

Given the back-order situation with the GP cameras it would be very helpful if we could have access to more footage if possible.
Cheers.

fbriggs, 2016-08-08:
Ok, here are a few tests on my machine..
Mac Pro, 3.5GHz 6-Core Intel Xeon E5, 64GB RAM 1866 MHz DDR3

ISP step (alone, 1 frame):
ISP runtime: 0:00:11.824214
(this has to be done once per frame)

Rectify step (alone):
RECTIFY runtime: 0:04:39.028184
(this is done once per camera or once per video, not once per frame)

Render step (alone, 1 frame). These are the options we use most often:
- quality = 6k
- optical flow algorithm = pixflow_low
- enable_top, enable_bottom, enable_pole_removal

RENDER runtime: 0:00:26.457989

Render step (same as above + enable_render_coloradjust)
- NOTE: that this option causes the first frame to be rendered twice- the first time it renders the first frame, the result is discarded, but it saves some data that helps it fix colors. After that, the color adjustments are applied to all subsequent frames.
- NOTE: in addition to the one-time cost, this will also add about 5 seconds per frame to render time.
- NOTE: enable --verbose, and look in the "Runtime breakdown" section of the logs to see more info on runtime for rendering each frame. 

Time to render second frame (after color-adjustment profile is built): 32.1149 sec

So looking at the time for steps that must be done every frame, the time to render a video is dominated by
ISP step - 11.82 sec/frame
RENDER step - 32.11 sec/frame or 26.45 sec/frame (with or without automatic color correction)

fbriggs, 2016-08-08:
Your runtime of ~14 min for rectify seems suspiciously high.

closed 2016-08-08 by fbriggs

VRabbitHole, 2016-08-08:
Very helpful - thanks!

And yes, agree that there's something 'odd' about the dual Xeon proc machine's performance .. it's my go-to machine for traditional post VR workflow using Kolor stitching and PremPro/Mettle with Resolve color correction (with 2 x Titans or W9100s) but for FB optical-flow computation looks like single, fast i7 is the way to go .... 5 minutes for the RECTIFICATION processing is much more reasonable.

Looks like your render speeds are similar to OZO times on a Mac Pro - good to know but would expect better on a new SkyLake with DDR4 ... just wondering why you stick with the default 'quality = 6k' setting when max resolution of .mp4 is 4096 x 2048 ... Nyquist maybe?

Also, what's your thought on rendering out HEVC/H265 at the end instead of H264?

fbriggs, 2016-08-08:
It is true that the maximum resolution that can be decoded by most current-generation video decoding chips is near 4096x2048. However, Facebook has a technology called "viewport adaptive streaming" which allows us to deliver higher resolution, by encoding many different copies of the video, each of which is high res in one direction, and lower res in the opposite direction. As the user moves their head, which version they get is switched dynamically. Using this approach, we can actually play 6k. Currently this is available only through Oculus Video, and only in use on a relatively small number of videos.

In the future, we hope more video players support 6k or 8k playback directly.

Our rendering software also supports output of cubemap format instead of equirect, which gives better quality in VR for the same pixel resolution. However, internally, the renderer always uses equirect, then maybe converts to cubemap at the end. For example, you could specify the following options to produce the highest quality photo that can be sideloaded into Oculus 360 Photos (not video):

- quality: 8k
- cubemap face resolution: 1536
- cubemap format: photo

fbriggs, 2016-08-08:
Regarding h264 vs h265 - we currently don't use h265 for legal/license related reasons. Oculus Video is capable of playing h265.

fbriggs, 2016-08-08:
Oh, one more thing I forgot to mention- if you render an h264 mp4 at 6k quality, then upload it to Facebook, our servers will do the "viewport adaptive" encoding, and users can find it on Facebook, then play it in Oculus Video at the full 6k quality (on GearVR, not Rift). Currently this is the only way to play 6k quality stereo video in VR that we are aware of by any means. This functionality is relatively new, so if you try it and it doesn't work as expected please post an issue here.

VRabbitHole, 2016-08-08:
Brilliant .... really impressed with how you guys have thought through the complete end-to-end acquisition/post/delivery pipeline ... mono 360º capture and mobile deliver platform issues are the Achilles heel of wide scale non-gaming VR adoption at the moment ... I gave a presentation at VRLA on Saturday on the importance of starting at the end and working backwards from the HMD to the acquisition device (also gave a shout out to the FB 360 Surround camera - said it would move the VR goal posts ... lots of quizzical looks) ... all that matters to Joe Public is the emotional experience they get in the HMD and whether they have a pleasant and palpable sense of presence.

Where can I find out more about FB 'viewport adaptive encoding' architecture for mobile? Is the 360 camera team driving that dev effort as well? ... want to try out 6K encoding and see how it plays in GearVR.

fbriggs, 2016-08-08:
Here's a FB blog post about viewport adaptive streaming from a little while ago:
https://code.facebook.com/posts/1126354007399553/next-generation-video-encoding-techniques-for-360-video-and-vr/

This may not be completely up-to-date (i.e. we have moved on to better approaches). It is a different team at Facbeook that works on this, but the camera team is in close contact.