facebookarchive/Surround360 #40: Stereo inconsistency

mimiaofandisp, 2016-08-08:
I generated the panorama using the sample data, but the result has a lot of stereo inconsistency, both in geometric structure and color. I am not sure if I got the wrong result or your algorithm has issues in handling stereo consistency. Could you please provide your results as a ground truth? Thank you!

fbriggs, 2016-08-08:
We recommend using the python script 'run_all.py'. To run it, cd to /surround360_render, and run 'python scripts/run_all.py'. From here you should have several options:

* To fix color inconsistency, check "enable_render_coloradjust".
* To fix geometry inconsistency (vertical parallax), make sure you have run the 'rectify' step in the list of steps.

mimiaofandisp, 2016-08-08:
For geometry inconsistency, there is no vertical parallax, but the objects shapes of left and right view are inconsistent. I did rectification with your config files.

fbriggs, 2016-08-08:
Here is a sample result, along with the settings that were used to generate it:
https://s3-us-west-2.amazonaws.com/surround360/sample/sample_result.zip

fbriggs, 2016-08-08:
Other inconsistency may be caused by optical flow error. If you post the result you are getting, I can look at tell you if it is a know/expected issue or not.

mimiaofandisp, 2016-08-08:
Thanks for the sample output, it is much better than what I got:
[image]
[image]

fbriggs, 2016-08-08:
You results look similar to what I expect. I suggest enabling the following as well:
enable_top
enable_bottom
enable_poleremoval
enable_render_coloradjust

mimiaofandisp, 2016-08-08:
Thanks for the reply. But is there a reason why the stereo consistency
cannot be well maintained without enabling top and bottom camera?

mimiaofandisp, 2016-08-08:
I just enabled top and bottom camera, the result did not change too much. Still too much stereo inconsistency:

[image]
[image]

fbriggs, 2016-08-08:
In the result you posted, which type of stereo inconsistency are you seeing? Can you draw a red circle around it and post that?

If the issue is color, then try running with enable_render_coloradjust

mimiaofandisp, 2016-08-08:
See the blue box with the white cover? its shape is completely different in the left and right view

fbriggs, 2016-08-08:
Ok, I see what you are talking about now. Interestingly, I did not get the same result on this dataset (see https://s3-us-west-2.amazonaws.com/surround360/sample/sample_result.zip).

Did you change any of these settings:
- optical flow algorithm: default is "pixflow_low"
- any changes to config files?

mimiaofandisp, 2016-08-08:
I did not change the optical flow config
I changed these to parameters
DEFINE_int32(eqr_width,                   256,            "height of spherical projection image (0 to 2pi)");
DEFINE_int32(eqr_height,                  128,            "height of spherical projection image (0 to pi)");

As i got the error message eqr_width should be 14x, so i changed them to 3500 and 1750, though i really do not how to setup these two numbers.

mimiaofandisp, 2016-08-08:
i tried different numbers, small values gave really low quality output, so changed them to large number.

fbriggs, 2016-08-08:
Ok thanks. It looks like you are directly running bin/TestRenderStereoPanorama. That is OK, but it may be easier/more convenient to run 'python scripts/run_all.py'. This is a higher level wrapper around TestRenderStereoPanorama and other tools, and provides a GUI to config it.

The results that I posted are using "6k" quality defined in run_all.py, which calls TestRenderStereoPanorama passing in eqr_width=6300 and eqr_height=3072.

This may explain some of the differences. (Usually) the higher resolution you go, the better the results will be for optical flow. The highest resolution you can use to get a benefit is 8k, which is eqr_width=8400, eqr_height=4096.

fbriggs, 2016-08-08:
Also in https://s3-us-west-2.amazonaws.com/surround360/sample/sample_result.zip, there is a screenshot of the GUI showing exactly how I configured it to get the result.

mimiaofandisp, 2016-08-08:
Thanks. so how do you determine the ratio for different images? what if i use my own images? still 6300 and 3072?

fbriggs, 2016-08-08:
There are two different sets of parameters:
eqr_width and eqr_height
final_eqr_width and final_eqr_height

The "final" version are what will be output at the end (it rescales). So these should be set based on what video resolution you can play or what resolution you want to edit for post-production.

The eqr_width and eqr_height parameters are used internally. eqr_width must always be evenly divisible by the number of cameras on the side. eqr_height should be close to eqr_width/2.

You can specify any value you want for eqr_width and eqr_height, but the highest eqr_width you can specify without ever up-scaling an image in the pipeline is around 8000. Higher than that and the pipeline will be operating on blurry scaled up versions of the input, which doesn't help.

closed 2016-08-08 by fbriggs

mimiaofandisp, 2016-08-08:
Thanks for the clarification. However, i just finished running the program
with higher resolution, the stereo inconsistency is still there. Any idea
why?

fbriggs, 2016-08-08:
Try running again using 'python scripts/run_all.py' and make sure your configs exactly match those in sample_result.zip.

In particular, make sure you have enable_render_coloradjust turned on. This affects colors, and may have some affect on flow.

mimiaofandisp, 2016-08-09:
well, I tried different ways and the results are the same. I actually split your sample result into left and right view and found that it has the same problem.