facebookarchive/Surround360 #42: Intermediate results for optical flow testing

mimiaofandisp, 2016-08-09:
Hi, I wonder if you could also provide the intermediate results generated from optical flow, cause I got some really weird novel view results. Are these what I supposed to get?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

fbriggs, 2016-08-09:
These definitely look messed up. My guess is that you are trying to input projections into flow, but that is actually not what goes into flow (it is the overlap images, found in /flow_images that go into flow). I can't really do much unless you provide complete information about what you are doing, i.e. all commands you are running + anything else you did.

My general recommendation is to use 'python scripts/run_all.py' and make sure your UI configuration exactly matches the screenshot that goes with sample_result.zip.

closed 2016-08-09 by fbriggs

mimiaofandisp, 2016-08-09:
OK, that's the reason, I input the projections into flow. 
Yeah, I already used run_all.py and got the exact same result as your sample output.
trying to figure out the algorithm now.