facebookarchive/Surround360 #47: Fatal Error on compiling CameraControl Software

VRabbitHole, 2016-08-12:
Just went through the install procedure for the CameraControl software and in the final compile section get this Fatal Error Message .... any suggestions on where I might have gone wrong (I'm not a C++ or C guy):

fbvr22@FBVR22:/Surround360/surround360_camera_ctl$ make
[ 40%] Building CXX object CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o
/Surround360/surround360_camera_ctl/source/camera_control/CameraControl.cpp:49:32: fatal error: libavcodec/avcodec.h: No such file or directory
compilation terminated.
CMakeFiles/CameraControl.dir/build.make:86: recipe for target 'CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o' failed
make[2]: *** [CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/CameraControl.dir/all' failed
make[1]: *** [CMakeFiles/CameraControl.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
fbvr22@FBVR22:/Surround360/surround360_camera_ctl$ 

Here's the ls -l output:

fbvr22@FBVR22:/Surround360/surround360_camera_ctl$ ls -l
total 68
drwxr-xr-x 2 root root  4096 Aug 12 10:28 bin
-rw-r--r-- 1 root root 10486 Aug 12 10:28 CMakeCache.txt
drwxr-xr-x 8 root root  4096 Aug 12 10:29 CMakeFiles
-rw-r--r-- 1 root root  1388 Aug 12 10:28 cmake_install.cmake
-rwxr-xr-x 1 root root  1377 Aug 12 09:20 CMakeLists.txt
drwxr-xr-x 2 root root  4096 Aug 12 09:20 configs
-rw-r--r-- 1 root root  1550 Aug 12 09:20 CONTRIBUTING.md
-rw-r--r-- 1 root root  1531 Aug 12 09:20 LICENSE_camera_ctl.md
-rw-r--r-- 1 root root  8692 Aug 12 10:28 Makefile
-rw-r--r-- 1 root root  4106 Aug 12 09:20 README.md
drwxr-xr-x 2 root root  4096 Aug 12 09:20 scripts
drwxr-xr-x 4 root root  4096 Aug 12 09:20 source

ted51, 2016-08-15:
I met this error before and fixed it by changing the way to compile ffmpeg:

 

./configure --enable_shared

make

sudo make install

 

 

Best regards,

 

Ted Gao(高昌军)

--------------------------------------

Chengdu CK Science & Technology

MP : +86 15008464587

TEL: +86 028 68616921

Email: changjun.gao@ck-telecom.com <mailto:changjun.gao@ck-telecom.com> 

--------------------------------------

 

发件人: noreply@github.com [mailto:noreply@github.com] 代表 VRabbitHole
发送时间: 2016年8月13日 1:59
收件人: facebook/Surround360
主题: [facebook/Surround360] Fatal Error on compiling CameraControl Software (#47)

 

Just went through the install procedure for the CameraControl software and in the final compile section get this Fatal Error Message .... any suggestions on where I might have gone wrong (I'm not a C++ or C guy):

fbvr22@FBVR22:/Surround360/surround360_camera_ctl$ make
[ 40%] Building CXX object CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o
/Surround360/surround360_camera_ctl/source/camera_control/CameraControl.cpp:49:32: fatal error: libavcodec/avcodec.h: No such file or directory
compilation terminated.
CMakeFiles/CameraControl.dir/build.make:86: recipe for target 'CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o' failed
make[2]: *** [CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/CameraControl.dir/all' failed
make[1]: *** [CMakeFiles/CameraControl.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
fbvr22@FBVR22:/Surround360/surround360_camera_ctl$ 

Here's the ls -l output:

fbvr22@FBVR22:/Surround360/surround360_camera_ctl$ ls -l
total 68
drwxr-xr-x 2 root root 4096 Aug 12 10:28 bin
-rw-r--r-- 1 root root 10486 Aug 12 10:28 CMakeCache.txt
drwxr-xr-x 8 root root 4096 Aug 12 10:29 CMakeFiles
-rw-r--r-- 1 root root 1388 Aug 12 10:28 cmake_install.cmake
-rwxr-xr-x 1 root root 1377 Aug 12 09:20 CMakeLists.txt
drwxr-xr-x 2 root root 4096 Aug 12 09:20 configs
-rw-r--r-- 1 root root 1550 Aug 12 09:20 CONTRIBUTING.md
-rw-r--r-- 1 root root 1531 Aug 12 09:20 LICENSE_camera_ctl.md
-rw-r--r-- 1 root root 8692 Aug 12 10:28 Makefile
-rw-r--r-- 1 root root 4106 Aug 12 09:20 README.md
drwxr-xr-x 2 root root 4096 Aug 12 09:20 scripts
drwxr-xr-x 4 root root 4096 Aug 12 09:20 source

VRabbitHole, 2016-08-15:
Thanks Ted ... will give it a whirl.

Have you actually built the 360 camera rig or just testing the software?

IsaacZhi, 2016-08-23:
I had built the code.

First, you need compile share lib ( .so files ) for ffmpeg.
In the guide ( https://trac.ffmpeg.org/wiki/CompilationGuide/Ubuntu  ), it only guide you to compile the static lib ( .a files ) , you need add --enable-shared in ffmpeg and libx265 and etc.

Second, if you install ffmpeg to your local path, you need add the include path and lib path in CMakeList.txt .

rezanvshield, 2017-01-20:
> _Second, if you install ffmpeg to your local path, you need add the include path and lib path in CMakeList.txt_ 

Thank you, but could you please explain how exactly this is done?

aparrapo, 2017-02-19:
@rezanvshield, assuming you have ffmpeg installed in <ffmpeg_path> you can do:

```
export LD_LIBRARY_PATH=<ffmpeg_path>/lib/
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:<ffmpeg_path>/lib/pkgconfig
export PKG_CONFIG_LIBDIR=$PKG_CONFIG_LIBDIR:<ffmpeg_path>/lib/
```

and run the cmake command again.

aparrapo, 2017-04-22:
@rezanvshield closing this task for now. Please feel free to reopen it if the problem persists.

closed 2017-04-22 by aparrapo