facebookarchive/Surround360 #48: Ubuntu 16.04 LTS?

VRabbitHole, 2016-08-12:
Just did a clean install of Ubuntu 16,04 LTS on my dual Xeon proc machine .... updated a few of the libs and GreyPoint has drivers for Ub16.04 LTS ... any reason to believe that the Camera Control software will not function as designed on 16.04 (assuming I can get the final compile to work - see Issue #6.

IsaacZhi, 2016-08-23:
Have you read the README file in Point Grey's SDK flycapture2-2.10.3.169-amd64-pkg.tgz ?
In the README file, Point Grey said that,

Strongly recommended:
Ubuntu 14.04.2 LTS: Linux kernel version 3.16.0-60 or later.
Ubuntu 14.04.3 LTS: Linux kernel version 3.19.0-51 or later.
Ubuntu 14.04.4 LTS: Linux kernel version 4.2.0-35 or later.   (*)
Ubuntu 16.04 LTS:   Linux kernel version 4.4.19-35 or later.  (*)

(*) On kernel series 4.2.X and later, issues have been encountered with Link
    Power Management (LPM) and USB3.0 devices on Linux. To resolve this, please
    contact Point Grey Support for assistance with compiling a customized kernel
    which resolves this issue.

I think if you use the Ubuntu 16.04 , you need to compiling a customized kernel.

mybuddyhoward, 2016-10-20:
@VRabbitHole You can use Ubuntu 16.04 as long as your kernel version is 4.4.0-25 or later. The LPM issue on Intel USB3 devices has been resolved if you used that or a later version. This information was provided to us from Point Grey support. Hope this helps.

closed 2017-02-19 by aparrapo