facebookarchive/Surround360 #49: Single capture box approach?

VRabbitHole, 2016-08-12:
My dual proc Xeon box has 8 x PCIe slots so I'm thinking of putting the 5 x USB cards in there plus a SAS HBA plus a 10Gbs Nic.

Then mounting the box on a custom built wheeled dolly with the camera tripod secured to it along with the UPS .... run the 17x USB cables directly into the box and store the live data directly onto the 16 x SSDs inside the box .... then run two Ethernet cables to my i7 render machine (plus the power cable to the UPS) .... one for the IPMI/KVM and the other for the 10Gbs fie transfer to the i7.

My rationale for this approach is this:

1) I only need to have one box on set directly under the camera ... it's a heavy SuperMicro chassis so will be stable even on wheels.
2) I only need to run 3 x black cables (robust, easily sourced and cheap) from the camera rig to the DIT station (Digital Imaging Technician).
3) the DIT will have the live data on the Xeon machine under the camera (ZFS or BTRFS equivalent of RAID 5) and a back-up set on the i7 .... with 10Gbs NFS we can pull data off the Xeon box at around 900 MB/s. DITs always make at least one back-up copy of the camera neg on set anyway and this way we could also make another secure copy to an encrypted portable SSD if needs be off the i7 data pool.
4) while the Xeon box is securely capturing data the DIT can be rendering the RAW files on his i7 machine with decent monitors attached plus HMDs.

Any reason or concerns on this approach?

I totally get the thinking behind the LunchBox/USB Expansion Chassis/Areca 8 bay approach if you're out in the field or a live concert situation but on a Sound Stage or controlled environment, a single, heavy box under the camera rig with three long, durable, bog-standard cables running off it, might be a better alternative.

Appreciate feedback on the idea and any other approaches being considered by folks.

bkcabral, 2016-08-12:
We initially started with the single capture "box" approach.  So we know it works as long as you have enough cross-sectional PCIe BW and enough processors so that the multi-threaded capture SW won't hiccup.   We went to a two box solution because it's easier to post out something small under the camera than big

VRabbitHole, 2016-08-12:
Thanks for the feedback ... and yes, I'm still in two minds about a single big box versus three smaller boxes ... pros and cons either way .... I think the dual proc Xeon should have enough BW ... I've used it with 4 x W9100s as a DaVinci Resolve render machine ... can handle realtime debayer of 6k Dragon .r3ds. 

The thing that appeals to me is having everything on a small dolly .... assuming it's rock solid with no vibrations being transmitted up the pole to the cameras, it could be an easy and quick way to move the whole 'kit and caboodle' around a set or concert area with only three trailing cables. My main concern at the moment is the noise level under the camera ... the SuperMicro fans are noisy as hell ... if the single box approach is viable will probably have to swap out the OEM fans for quiet SilenX ones. 

Have you tested your rig on a dolly with lateral movement or are you assuming peeps will only use it on lock-down with jump cuts between takes?

VRabbitHole, 2016-08-12:
Re the 'easier to post out something small under the rig' observation, what's the max dimensions you think would work? .... eg, the SMC chassis is 17.5 ins x 27 ins 4U workstation.

I presume for post rig removal you were using the FB built in 'post removal' module or were you using NUKE?

fbriggs, 2016-08-12:
I don't have a number for max dimension, but it is critical to keep anything you want to post/plate out inside the monoscopic region at the bottom.

VRabbitHole, 2016-08-12:
How can we calculate that max circumference in relation to the distance between the bottom of the camera to the top of the box? ... now I wish I'd paid attention in my spherical trig classes!

fbriggs, 2016-08-12:
To make things easy, assume the side cameras have 90 degree vertical FOV (its actually ~77 after correcting barrel distortion, but assuming 90 will give you a more conservative estimate).

In this case, you just draw a 45 degree line down from the camera to the ground and that marks the safe zone. In other words, if the tripod height is H, then the circle of safety has radius H.

In practice, I would try to keep things much smaller than that... you shouldn't need a 5 foot radius pile of equipment.

VRabbitHole, 2016-08-12:
Interesting ... so if we have the mid-point horizontal plane of the flying saucer set at around eye-level, say 5ft 6in depending on the talent and the show, then the 32 inch diagonal of the SMC box would easily fit inside say a 48 in radius 4 ft below ?

If this schematic is drawn anywhere to scale then the 4 x box combo looks like it's taking up as much room (if not more) as the single big box approach ... plus it's not mobile:

[image]

shanjay25, 2016-09-01:
To be clear in what I'm thinking ... I'd mount the camera rig tripod directly on top of the SMC box with the UPS snuck in there as well, ie, not have the tripod or UPS touching the ground but secured to the dolly/SMC box somehow (tbd).