facebookarchive/Surround360 #51: 360 Surround Camera Rental in LA area?

VRabbitHole, 2016-08-12:
Has anyone in LA area built the 360 beast?  Any rental packages available?

Anyone with a camera rig interested in collaborating on end-to-end workflow testing?

Brian, I don't suppose there's any chance that FB would let us book some time in your lab to let us shoot some test footage of our own on one of your rigs? ... bring our Capture and Render boxes up to Menlo Park to record some charts and people moving around ... I know it's a total pain in the butt to organize - I run the VR Lab @ LumaForge in Hollywood, so believe me I know ... but with the next batch of PG cameras not shipping until the end of Oct we're going to be snookered on testing the workflow in its entirety for another couple of months now.

Hugomalencar, 2016-08-19:
Hey VrabbitHole,

what exactly are you trying to produce? just some test footage of the facebook camera rig? 

I work for an entertainment company based in culver city doing 360 and we have cameras that we could rent it out along with our services. 

Hit me up with an email to hugo@wearefamous.com  if your interested so we can talk better and you can explain your project better. more than welcome to come over to our facility and check it out.

cheers

VRabbitHole, 2016-08-19:
Do you have the Facebook 360 Surround camera in stock and available now?

Hugomalencar, 2016-08-19:
currently not the facebook, but we might look into it...

we have other high quality cameras such as the ozo

VRabbitHole, 2016-08-19:
Two quick observations ... (1) this github page is dedicated to the new FB 360 Surround camera - that's what I was specifically asking about ... (2) if you think the ozo camera is 'high quality' you're definitely on the wrong page.