facebookarchive/Surround360 #60: Data Directory location

knottb, 2016-08-17:
I am trying to process the sample dataset to learn how to use the run_all.py GUI and I can't figure out where the Data Directory should be. I have downloaded the sample_dataset but none of its subdirectories function correctly as the Data Directory (I don't see any .bin files and I don't understand this instruction)

fbriggs, 2016-08-17:
The sample dataset shows the state of the destination directory after already running the 'unpack', 'arrange', and 'isp' steps. To start at the beginning, you need the raw binary files that are captured by our camera control software. We haven't made any of those available at the moment.

To render the sample dataset, you should only run the 'render' step, or the 'isp' and 'render' steps. I have attached a short video that shows how to do this (without enabling all of the other options like top/bottom cameras).

[sample_dataset_minimal.mov.zip](https://github.com/facebook/Surround360/files/423572/sample_dataset_minimal.mov.zip)

closed 2016-08-17 by fbriggs

knottb, 2016-08-17:
You're right. I noticed that shortly after asking. Thanks.

Incidentally, my run_all.py does not recognize having a "rectify" step input. Am I using an outdated version? (It only has [unpack,arrange,isp,render,ffmpeg,all] options)

fbriggs, 2016-08-17:
yes, if you do not have the rectify step, then you have an old version. thanks!