facebookarchive/Surround360 #61: Compile Render part

EnriqueMVega, 2016-08-17:
Hi there
Do you plan to provide instructions to compile the render part on windows? Are there any plans to support the whole workflow under windows?
Thanks

fbriggs, 2016-08-17:
We do not provide instructions on how to compile for windows at this time. Most likely you would need to install cygwin first, then the cmake stuff might just work. If you get it working, please let us know.

closed 2016-08-17 by fbriggs

bkcabral, 2016-08-17:
I estimate a 2 to 5 days of work to make it work.  Nothing substantive - in terms of code changes other than to make the compiler happy.  We currently don't have windows plans but as Forrest suggests would be happy if someone wanted to take it on :-)