facebookarchive/Surround360 #62: Camera Control soft

BroadDong, 2016-08-18:
[image]
**The error is that there isn't "#include <libavcodec/avcodec.h>". But I have installed 
it. Why?**
[image]

superdog8, 2016-08-18:
Please configure ffmpeg with following flags:
 --enable-nonfree --enable-gpl --enable-libx264 --enable-shared --enable-libvpx

DxgHandInHand, 2016-08-29:
I have installed ffmpeg and configure it as show below:
--enable-nonfree --enable-gpl --enable-libx264 --enable-shared --enable-libvpx
But it failed. Why?
[image]

DxgHandInHand, 2016-08-29:
could you tell me the detail?

IsaacZhi, 2016-09-01:
You should add the ffmpeg's include path to gcc search path.

aparrapo, 2017-02-19:
@DxgHandInHand did you solve this issue?

aparrapo, 2017-04-22:
@DxgHandInHand closing this issue for now. Please feel free to reopen it if you still haven't solved it.

closed 2017-04-22 by aparrapo