facebookarchive/Surround360 #66: surround360_render not working

suhitpathak, 2016-08-22:
I am interested in testing the rendering software without first building the camera.
I am trying to use the sample dataset provided.
When i execute the command "python run_all.py", and enter the required data in the GUI, I get the following error. Can someone tell me what might be wrong?

_python /home/Surround360/Surround360-master/surround360_render/scripts/run_all.py --rectify_file NONE --src_intrinsic_param_file /home/Surround360/Surround360-master/surround360_render/res/config/sunex_intrinsic.xml --data_dir /home/Surround360/dataset/sample_dataset/sample_dataset --start_frame 0 --frame_count 0 --cubemap_format video --steps all --rig_json_file /home/Surround360/Surround360-master/surround360_render/res/config/17cmosis_default.json --dest_dir /home/Surround360/output --pole_masks_dir /home/Surround360/Surround360-master/surround360_render/res/pole_masks --quality 6k --cubemap_face_resolution 0 --cam_to_isp_config_file /home/Surround360/Surround360-master/surround360_render/res/config/isp/cam_to_isp_config.json --flow_alg pixflow_low

** UNPACK ** [Step 1 of 6]

E0822 16:05:58.043388  2313 SystemUtil.cpp:48] Terminated with VrCamException: file read failed:/home/Surround360/dataset/sample_dataset/sample_dataset/cameranames.txt
Aborted (core dumped)

UNPACK runtime: 0:00:00.066530

** ARRANGE ** [Step 2 of 6]

mkdir: cannot create directory `/home/Surround360/output/vid': File exists
mkdir: cannot create directory `/home/Surround360/output/eqr_frames': File exists
mkdir: cannot create directory `/home/Surround360/output/cube_frames': File exists
mkdir: cannot create directory `/home/Surround360/output/pole_masks': File exists
mkdir: cannot create directory `/home/Surround360/output/single_cam': File exists
Traceback (most recent call last):
  File "/home/Surround360/Surround360-master/surround360_render/scripts/arrange_dataset.py", line 39, in <module>
    file_list = remove_txt_files(list_only_files(src_dir))
  File "/home/Surround360/Surround360-master/surround360_render/scripts/arrange_dataset.py", line 8, in list_only_files
    def list_only_files(src_dir): return filter(lambda f: f[0] != ".", [f for f in listdir(src_dir) if isfile(join(src_dir, f))])
OSError: [Errno 2] No such file or directory: '/home/Surround360/output/raw'

ARRANGE runtime: 0:00:00.080141

** ISP ** [Step 3 of 6]

ISP runtime: 0:00:00.146664

    The 'rectify' step is enabled. The 'rectification file' parameter must not
    be NONE. Please supply a destination path to write the rectification file.

    A good path to use is your destination directory /rectify.yml._

fbriggs, 2016-08-22:
For the sample data, we are giving you the state after running the 'unpack', 'arrange', and 'isp' steps. The error message you are getting comes from it trying to run 'unpack', but it doesn't have the packed version of the data.

To get it to work with the provided sample data, only run the 'render' step.

fbriggs, 2016-08-22:
Here's a short video showing how to do it:
https://github.com/facebook/Surround360/files/423572/sample_dataset_minimal.mov.zip

closed 2016-08-22 by fbriggs