facebookarchive/Surround360 #68: json file configuration

mimiaofandisp, 2016-08-23:
Hi, i want to use my own camera array, for the configuration
if I use gopro cameras like google jump, what the numbers should i use? the aspect ration is 3:4, horizontal fov is 94.4, vertical is 122.6

fov_horizontal: 94.4*0.85??
aspect_ratio_wh: 3/4??

fbriggs, 2016-08-23:
The value of 0.85 is specific to the Sunex lenses we are using and will generally not be applicable to GoPro lenses. TO get the proper fov_horizontal value to use, you need to run the intrinsic calibration procedure, and it prints the value at the end. Refer to the section "Intrinsic Calibration" in README.md.

The aspect_ratio_wh is the width / height of the image in pixels.

closed 2016-08-23 by fbriggs

mimiaofandisp, 2016-08-23:
Do you mean the estimatedFovX estimatedFovY and estimatedAspectRatio?

fbriggs, 2016-08-23:
estimateFovX is what you should use for fov_horizontal.
ignore estimatedAspect ratio, and just use the ratio of the image width / image height.

fbriggs, 2016-08-23:
I am curious how things work with GoPro. is you have any datasets / results to share, that would be very interesting, and I might have more suggestions to help. Thanks for you work on this.

mimiaofandisp, 2016-08-23:
sure. thank you.