facebookarchive/Surround360 #71: Different camera option?

natevplas, 2016-08-25:
This is less an issue than just a question, since I have not attempted to assemble this.  
1. Why did Facebook choose the [Grasshopper camera](https://www.ptgrey.com/grasshopper3-41-mp-color-usb3-vision-cmosis-cmv4000-2-camera) from PointGrey rather than the Chameleon (specifically [CM3-U3-50S5C-CS](https://www.ptgrey.com/chameleon3-50-mp-color-usb3-vision-sony-imx264-2))?  The Chameleon has a much better dynamic range (70.6 dB vs 53.19), is more than $500 cheaper, higher resolution, and a lot smaller.  The only thing better about the Grasshopper is the frame rate.
2. Could I change the rig in order to accommodate the Chameleon and have it still work with the stitching software?  I'm unclear on how flexible the stitching software is.
3. How in the world do I find videos that have been shot with the Surround360 on the GearVR?  I'm very curious to see the stereoscopic effect in motion (I already downloaded the sample still from here).

If anyone has an answer to one or more of these questions, please let me know!

aparrapo, 2016-09-07:
1. It was the frame rate mostly. 35 fps is usually not good enough for most of the shots, and we want to be able to do at least 60 fps.
2. Yes, you can replace the Grasshoppers with Chameleon and the stitching software will work just fine. You can theoretically use any camera you want as long as they are positioned in the side + top/bottom fashion. If you change the number of side cameras or the ring radius you just have to modify the config file in github.com/facebook/Surround360/blob/master/surround360_render/res/config/17cmosis_default.json
3. We will probably upload a video soon :)

natevplas, 2016-09-07:
Gotcha.  
2. Looks surprisingly easy to edit the code, but also looks like if you did change cameras or lenses you would have to correct some of the FOV and resolution numbers as well.  **What would happen if you changed the orientation of the side cameras (rotate them 90 degrees)?**  The Chameleon has its USB port on the side of the camera, which would get in the way.  If you rotated the camera, you could have the port on the top or bottom.
3.  Let me know when you do!  Does anyone else know where I can see some footage in the meantime?

aparrapo, 2016-09-07:
2. Correct. If you do and have any problems just let us know, we will be more than happy to help you :) If you rotate the Chameleon 90 degrees you will lose some effective FOV in one of the directions, although given that the Grasshoppers are 2048x2048 and the Chameleon are 2448x2048, if you use 17 side cameras you will be ok. You will just have to rotate the images you get from the cameras, as they will probably be sideways :)

closed 2016-09-12 by fbriggs