facebookarchive/Surround360 #74: number of cameras and overlap

cvaldesmol, 2016-08-29:
I have been playing with i = r * sin( FOV/2 - 360 / n)
My cameras have a vfov of 79 after distortion correction. I see I can work with 6 cameras (maybe 5 if I increase the r) but what about overlap? how do you take that into account? what is recommended?
with 6 cameras I get 79*6 = 474 degree
I heard that you need at least 720 degree if you want 3d, is that true?

fbriggs, 2016-08-29:
With our software, you need to have i > 3.2cm. There is nothing about our software that relates to 720 degrees. 6 cameras is probably not enough, unless you have a very large r (which won't work well because it makes the optical flow problems too hard). For example our current rig has n = 14, FOV = 77, and r = 21cm.

closed 2016-08-29 by fbriggs

cvaldesmol, 2016-08-29:
I kept thinking and I guess that to get 3d you need full overlap of each camera. Maybe something like camera 1 overlaps 50% of camera 2, and camera 2 overlaps 50% of camera 3. So camera 2 has full overlap using camera  1 and camera 2. So for my 79 degree camera I need 10 cameras regardless of r. Am I right?
If that is the case, is 10 good enough to get high quality 3d? or more overlap is required ?

fbriggs, 2016-08-29:
It has nothing to do with how many overlaps or how much overlap you have. Its all about that formula i = r * sin(FOV/2 - 360/n).

10 cameras might be enough, but I suggest at least 12.