facebookarchive/Surround360 #76: suggested value of r with gopro camera rig setup

ssaroha, 2016-08-30:
Hi,
I played around with the Surround360 sample data set to try out the rendering software. looks near. Next wanted to see if we can use this stitching software for our gopro setup.

We have a gopro camera rig setup as shown here: https://www.amazon.com/Degree-Panorama-Mount-GoPro-Accessory/dp/B01FR5W622

with total of 7 cameras with 5 in the ring, 1 on top and 1 on bottom (for supposed greater overlap) 

these gopros with ultra-wide settings have FOV of 170. Using the formula you have suggested, If 
we put n=5, FOV=170
i = r * sin(FOV/2 - 360/n)
i = r * Sin(85-72)
i = r * Sin(13)
i = 0.22 * r

r has to be > 14.5 cms to meet the i>3.2 cm target.

If we put n=7, then
i = r * Sin(170/2 - 360/7)
i = r * Sin(85 - 51.43)
i = r * 0.55

with n=7, r can be 5.8 cm, which seems more in line with whats possible with our current setup.
Couple of questions:

1. Can we take n=7 for calculation purposes with our rig setup https://www.amazon.com/Degree-Panorama-Mount-GoPro-Accessory/dp/B01FR5W622

2. If the setup is fine, is it possinle to use CameraControl software for generating the frames which would be used by camera render software.

fbriggs, 2016-08-30:
1. I'm not sure I understand your question. If you have n=5 cameras in a ring (as shown in the link), then you can't treat that like n=7. If you build a rig with radius 5.8cm and n=5 cameras, when you render at 6.4cm IPD, there will be black stripes in the panorama where it doesn't have enough data to make the right color.

2. No, the CameraControl software is specifically designed for interfacing with PointGrey cameras. However, what you can do is set up a directory structure the same way that run_all.py does, then put your GoPro images in the isp_out folders. After that is setup (possibly some scripting involved), you can render with our software.

closed 2016-08-30 by fbriggs

akc2267, 2016-10-04:
Also, fov is the HORIZONTAL fov. 170 degrees is the GoPro's diagonal fov.