facebookarchive/Surround360 #78: How to use the web to capture video ?

IsaacZhi, 2016-08-30:
I have compiled the CameraControl, and in /usr/local/bin/ make a soft link fir CameraControl ( I see the php file it use /usr/local/bin/CameraControl  )
I visit http://localhost/  
I have only one camera , I modify the capture_image.php let $num_cameras = 1;

When I press the "Preview" button , it will do some works, and show "Record" button, but after 5 seconds it will back to show the org web.  And I can not see the preview video.

I check the output dir, there is only a file cmd.txt
ubuntu:/media/snoraid/1472524227_1472522656$ cat cmd.txt
/usr/local/bin/CameraControl -fps 30 -numcams 1 -raw -nbits 8 -dir 1472524227_1472522656 -shutter 20 -gain 0

IsaacZhi, 2016-08-30:
I found that in php it call command like /usr/local/bin/CameraControl $cmd_options
And there no use sudo.

In Terminal, if I don't add sudo before call CameraControl, it will be terminated.
It report "terminate called after throwing an instance of 'boost::interprocess::interprocess_exception'
  what():  Permission denied
Aborted (core dumped) "

Maybe I should add sudo permission to php.

IsaacZhi, 2016-09-01:
We change the user "www-data" 's permission, to fix the problem.

closed 2016-09-01 by IsaacZhi

DxgHandInHand, 2016-09-26:
@IsaacZhi HI，could you tell me how the single camera caputures?