facebookarchive/Surround360 #80: Rectilinear to spherical

NinaMa17, 2016-08-31:
Hi, I'm trying to understand how you find the spherical coordinates from the rectilinear image. I am confused with the coordinate system, especially the 'f-coordinate'. At first I thought (u,v,f) is just a normal 3d coordinate system. But it seems like the value of f is constant because imageSize.width and fovH won't change no matter which point I choose. Also, I'm not sure what is the difference between theta and thetaU (phi and phiV). Can you explain this part a bit more??

shanjay25, 2016-09-01:
I'm still need learn more

aparrapo, 2016-09-01:
Hi @NinaMa17,

I've drawn a diagram to make it easier to understand (see image below).

`u` and `v` are coordinates on the image plane, and `f` is the focal length (which is fixed). To convert a rectilinear image to spherical we basically have to find the angles `theta` and `phi` in polar coordinates for a given `(u, v)`.

Now, `theta` is in the range [-FOVh/2, FOVh/2], and `phi` in [pi/2 - FOVv/2, pi/2 + FOVv/2], but on the projected image they both need to be in [0, W] and [0, H] respectively, where W and H are the width and the height of the destination image. `thetaU` and `phiV` are just rescaling these ranges to make it happen:

`thetaU = ((theta + fovH / 2.0f) / fovH) * float(imageSize.width - 1)`,

so when `theta = -fovH / 2`, `thetaU = 0`, and when `theta = fovH / 2`, `thetaU = W`. The same applies to `phiV`.

[image]

closed 2016-09-02 by aparrapo