facebookarchive/Surround360 #82: Compiling ffmpeg to include --enable-shared

s8i, 2016-09-01:
This seems to be a common issue when compiling a fresh version of ffmpeg
A step by step guide here would be very helpful if anyone has solved this on Ubuntu 14LTS or 16LTS

BroadDong, 2016-09-01:
When the ffmpeg is compiled,I encounter the issue as follow.Are they the same?
[image]

IsaacZhi, 2016-09-01:
If you need compiling the code add --enable-shared
libx265 
libvpx
ffmpeg
all need --enable-shared

superdog8, 2016-09-02:
@IsaacZhi is correct, all the dependencies require --enable-shared when compiling from source. Alternatively install from packages.

closed 2016-09-02 by superdog8