facebookarchive/Surround360 #83: The issue of installing the ffmpeg

BroadDong, 2016-09-01:
When I install the ffmpeg , it that is the error as follows always happened. Why? How may I do?
[image]

IsaacZhi, 2016-09-01:
What's the command you used?

Maybe you should first compiled the libvpx with --enable-shared

s8i, 2016-09-02:
This: https://github.com/facebook/Surround360/blob/master/README.md ask us to follow: https://trac.ffmpeg.org/wiki/CompilationGuide/Ubuntu

Unfortunately that guide does not give us a correctly compiled ffmpeg

BroadDong, 2016-09-02:
This is the step of installing and compiling the ffmpeg.Please tell me the error.
(1) 
sudo apt-get update
sudo apt-get -y install autoconf automake build-essential libass-dev libfreetype6-dev \
  libsdl1.2-dev libtheora-dev libtool libva-dev libvdpau-dev libvorbis-dev libxcb1-dev libxcb-shm0-dev \
  libxcb-xfixes0-dev pkg-config texinfo zlib1g-dev
mkdir ~/ffmpeg_sources
(2) yasm
cd ~/ffmpeg_sources
wget http://www.tortall.net/projects/yasm/releases/yasm-1.3.0.tar.gz
tar xzvf yasm-1.3.0.tar.gz
cd yasm-1.3.0
./configure --prefix="$HOME/ffmpeg_build" --bindir="$HOME/bin"
make
make install
make distclean
(3) libx264
cd ~/ffmpeg_sources
wget http://download.videolan.org/pub/x264/snapshots/last_x264.tar.bz2
tar xjvf last_x264.tar.bz2
cd x264-snapshot*
PATH="$HOME/bin:$PATH" ./configure --prefix="$HOME/ffmpeg_build" --bindir="$HOME/bin" --enable-static --disable-opencl --enable-shared
PATH="$HOME/bin:$PATH" make
make install
make distclean
(4) livbpx
cd ~/ffmpeg_sources
wget http://storage.googleapis.com/downloads.webmproject.org/releases/webm/libvpx-1.5.0.tar.bz2
tar xjvf libvpx-1.5.0.tar.bz2
cd libvpx-1.5.0
PATH="$HOME/bin:$PATH" ./configure --prefix="$HOME/ffmpeg_build" --disable-examples --disable-unit-tests --enable-shared
PATH="$HOME/bin:$PATH" make
make install
make clean
(5) ffmpeg
cd ~/ffmpeg_sources
wget http://ffmpeg.org/releases/ffmpeg-snapshot.tar.bz2
tar xjvf ffmpeg-snapshot.tar.bz2
cd ffmpeg
PATH="$HOME/bin:$PATH" PKG_CONFIG_PATH="$HOME/ffmpeg_build/lib/pkgconfig" ./configure \
  --prefix="$HOME/ffmpeg_build" \
  --pkg-config-flags="--static" \
  --extra-cflags="-I$HOME/ffmpeg_build/include" \
  --extra-ldflags="-L$HOME/ffmpeg_build/lib" \
  --bindir="$HOME/bin" \
  --enable-nonfree\
  --enable-gpl \
  --enable-libx264 \
  --enable-shared \
  --enable-libvpx 
PATH="$HOME/bin:$PATH" make
make install
make distclean
hash -r

s8i, 2016-09-04:
Brilliant, thanks for this! ...x265 not required?

s8i, 2016-09-05:
This compiles successfully on a fresh Ubuntu 16.04
*BUT* 
```bash
stage@stage:~/bin$ ffmpeg
ffmpeg: error while loading shared libraries: libavdevice.so.57: cannot open shared object file: No such file or directory
```
grrrr, frustrating : (

### install some dependencies  
`sudo apt-get update`
```bash
sudo apt-get -y install autoconf automake build-essential libass-dev libfreetype6-dev libsdl1.2-dev libtheora-dev libtool libva-dev libvdpau-dev libvorbis-dev libxcb1-dev libxcb-shm0-dev libxcb-xfixes0-dev pkg-config texinfo zlib1g-dev
mkdir ~/ffmpeg_sources
```

### yasm
```bash
cd ~/ffmpeg_sources
wget http://www.tortall.net/projects/yasm/releases/yasm-1.3.0.tar.gz
tar xzvf yasm-1.3.0.tar.gz
cd yasm-1.3.0
./configure --prefix="$HOME/ffmpeg_build" --bindir="$HOME/bin"
make
make install
make distclean
```

### libx264
```bash
cd ~/ffmpeg_sources
wget http://download.videolan.org/pub/x264/snapshots/last_x264.tar.bz2
tar xjvf last_x264.tar.bz2
cd x264-snapshot*
PATH="$HOME/bin:$PATH" ./configure --prefix="$HOME/ffmpeg_build" --bindir="$HOME/bin" --enable-static --disable-opencl --enable-shared
PATH="$HOME/bin:$PATH" make
make install
make distclean
```
### livbpx
```bash
cd ~/ffmpeg_sources
wget http://storage.googleapis.com/downloads.webmproject.org/releases/webm/libvpx-1.5.0.tar.bz2
tar xjvf libvpx-1.5.0.tar.bz2
cd libvpx-1.5.0
PATH="$HOME/bin:$PATH" ./configure --prefix="$HOME/ffmpeg_build" --disable-examples --disable-unit-tests --enable-shared
PATH="$HOME/bin:$PATH" make
make install
make clean
```
### ffmpeg
```bash
cd ~/ffmpeg_sources
wget http://ffmpeg.org/releases/ffmpeg-snapshot.tar.bz2
tar xjvf ffmpeg-snapshot.tar.bz2
cd ffmpeg
PATH="$HOME/bin:$PATH" PKG_CONFIG_PATH="$HOME/ffmpeg_build/lib/pkgconfig" ./configure \
--prefix="$HOME/ffmpeg_build" \
--pkg-config-flags="--static" \
--extra-cflags="-I$HOME/ffmpeg_build/include" \
--extra-ldflags="-L$HOME/ffmpeg_build/lib" \
--bindir="$HOME/bin" \
--enable-nonfree\
--enable-gpl \
--enable-libx264 \
--enable-shared \
--enable-libvpx 
PATH="$HOME/bin:$PATH" make
make install
make distclean
hash -r
```

IsaacZhi, 2016-09-05:
cd /etc/ld.so.conf.d/
sudo vim ffmpeg.conf
Add that:
# ffmpeg default configuration
/home/[your user name]/ffmpeg_build/lib

the path is your put the ffmpeg's lib path.

then
sudo ldconfig

DxgHandInHand, 2016-09-06:
@IsaacZhi ,您好，您是在北京吗？可以交流一下吗

IsaacZhi, 2016-09-06:
@DxgHandInHand , Yes, I live in Beijing. I have received your e-mail, I will reply you as soon as possible.

laisenglee, 2016-09-11:
你好，你在什么地方？我的邮箱：18688140876@163.com希望可以交流。我在广东我也在做这个项目，我已经把硬件部分做好了，软件部分基本也调试好了。 我的淘宝店铺里有Surround360 硬件部分在卖[https://item.taobao.com/item.htm?spm=a230r.1.14.277.SzXVpC&id=538209847817&ns=1&abbucket=17#detail]

dangsiper, 2016-09-19:
Is anyone live in Chengdu?

mybuddyhoward, 2016-09-21:
I'm struggling to get FFMPEG setup properly when I run 'make' on the camera.
My setup is Ubuntu 14.04 and I followed the compilation guide and this one but am encountering issues. Can someone help?

[ 33%] Building CXX object CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o
/home/howard/workspace/surround360/surround360_camera_ctl/source/camera_control/CameraControl.cpp:58:33: fatal error: libavutil/timestamp.h: No such file or directory
 #include <libavutil/timestamp.h>
                                 ^
compilation terminated.
make[2]: *** [CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o] Error 1
make[1]: *** [CMakeFiles/CameraControl.dir/all] Error 2
make: *** [all] Error 2

aparrapo, 2017-02-19:
ffmpeg has recently changed the installation steps and this should be fixed by now. Feel free to reopen it is the issue persists :)

closed 2017-02-19 by aparrapo