facebookarchive/Surround360 #93: Camera control

BroadDong, 2016-09-05:
When I run "./bin/CameraControl -numcams 17 -raw -nbits 8 -shutter 20 -gain 0 -debug", the error is bear as follow. Could you tell me the reason？And how I get the GUI of  the Camera Control?
[image]

mybuddyhoward, 2016-09-22:
I'm also seeing the same thing on my end. I've also ran that command under sudo but still the same results. Interested to see if there is a solution or reason for this?

malcolmbrown, 2016-12-09:
Hi. Did you find a solution to this? I had this section installed successfully and then installed everything else. Since then I get the same error.

malcolmbrown, 2016-12-12:
Hi. Did you find a solution to this? I had this section installed successfully and then installed everything else. Since then I get the same error. Using Sudo does not fix it.

malcolmbrown, 2016-12-13:
Hi. I solved this by mounting the RAID using the setup.sh in the scripts folder.

aparrapo, 2017-02-19:
Thanks for your help @malcolmbrown. Closing this issue.

closed 2017-02-19 by aparrapo