facebookarchive/Surround360 #94: GoPro 12 camera stitching setup

ssaroha, 2016-09-06:
I have 12 gopro 4 cameras in a rig setup, and was trying to use the surround 360 stitching software.
here are the details:
number of cameras = 12
Radius = 14 cm

Here is the zipped dataset, after I put the 12 camera images in "isp_out" folder.

https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/park_0905.zip

here is the script I used to run the software:
https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/run_park0905.sh

the stitching is not coming out right as shown by the generated image:
https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/eqr_000000.png

here is the output of the cmosis_default.json file:
{
  "camera_ring_radius": 14.0,
  "cameras":
  [
    {"camera_id": "cam0",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam1",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam2",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam3",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam4",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam5",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam6",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam7",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam8",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam9",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam10",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
    {"camera_id": "cam11",  "fov_horizontal": 120, "aspect_ratio_wh": 1.33},
  ]

I also tried to generate equi-rectangular photo with PtGui stitching software using the same set of 12 images, and the results are much more reasonable:
https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/ptgui-stitched-park.jpg

any ideas what tweaks we should be making to get it right?

thanks!

fbriggs, 2016-09-08:
You are on the right track as far as organizing the folder and putting the images in isp_out, and updating the rig json file. From looking at the command, I can see a couple of issues. I am including a copy of your command (reformatted) below:

```
./bin/TestRenderStereoPanorama
--logbuflevel -1
--log_dir /Users/satender/Downloads/s1/park_0905/logs
--stderrthreshold 0
--v 1
--src_intrinsic_param_file /Users/satender/Surround360/surround360_render/res/config/sunex_intrinsic.xml
--rig_json_file /Users/satender/Surround360/surround360_render/res/config/17cmosis_default.json
--ring_rectify_file NONE
--imgs_dir /Users/satender/Downloads/s1/park_0905/vid/000000/isp_out
--output_data_dir /Users/satender/Downloads/s1/park_0905/vid/000000
--prev_frame_data_dir NONE
--output_cubemap_path /Users/satender/Downloads/s1/park_0905/cube_frames/cube_000000.png
--output_equirect_path /Users/satender/Downloads/s1/park_0905/eqr_frames/eqr_000000.png
--cubemap_format photo
--side_flow_alg pixflow_low
--polar_flow_alg pixflow_low
--poleremoval_flow_alg pixflow_low
--cubemap_face_resolution 0
--eqr_width 3072
--eqr_height 1540
--final_eqr_width 3080
--final_eqr_height 3080
--interpupilary_dist 6.4
--zero_parallax_dist 10000
--sharpenning 0.25  
```

Here are all of the issues in the command you would need to fix to get a proper result:

`--src_intrinsic_param_file /Users/satender/Surround360/surround360_render/res/config/sunex_intrinsic.xml`
-->
This is passing it the intrinsic parameter for our lens, not the GoPro lens. The purpose of this calibration is to correct for barrel distortion, and it will be different on the GoPro lens. To fix this, you need to take a lot of images of a checkerboard with the GoPro, then run one of our tools, and change the xml file you pass in here. Check the "Intrinsic Calibration" section of /surround360_render/CALIBRATION.md

`--rig_json_file /Users/satender/Surround360/surround360_render/res/config/17cmosis_default.json`
-->
This is the path to our default rig json file for 14 cameras in a ring. If you modified that file like what you posted above, that is OK. Alternatively, you could make a different file and put your modifications in that, then change the path you pass in for --rig_json_file.

`--ring_rectify_file NONE`
-->
To get results that look good in stereo in VR, you should run the rectification step. Refer to CALIBRATION.md for more details.

Also, it would probably help make the color transitions between cameras smoother if you include this flag:
`--enable_render_coloradjust
`
You may have to runt he renderer twice to get the benefit (the first time it will create a calibration file for color, and the second time, it will use it).

One more tip:
You can use 'python scripts/run_all.py' to specify these parameters with a GUI instead of directly calling TestRenderStereoPanorama. If you do that, specify the destination directory as the directory that you included in the zip file. The "binary files" option there can be anything (it won't be used). Then specify only the steps you want to run (not 'all'. Probably 'rectify' and 'render').

I am closing this issue for now, but feel free to update it with any progress or further questions. Good luck!

closed 2016-09-08 by fbriggs

ssaroha, 2016-09-11:
thanks @fbriggs, the calibration step with checkerboard certainly helped!
using the generated intrinsic file for gopro, the stitching is much better, but i still see the different color shades across camera/stitching boundaries. 

here is the gopro intrinsic file generated by calibration step:
https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/my_intrinsic.xml

which I passed to the script: https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/run_park0905.sh

to generate following stitched image.
https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/eqr_000000_calibrated.png

most of the distortions are gone.

any ideas what further optimizations we could do remove boundary color differences.

Also for some reason the rectification step seems to make the output worse.
During the generation of rectify.yml file
I saw warnings like: 
"W0911 12:20:06.460093 2009489408 RingRectification.cpp:252] rectification objective is very bad. this suggests a problem. check /keypoint_vis for good matches"

here is the image with rectification step in the mix:
https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/eqr_000000_calibrated.png

here is the rectify.yml file:
https://dl.dropboxusercontent.com/u/1526357/gopro-stitching/rectify.yml

So effectively 2 questions:
- any way to reduce the color changes on the boundaries of the stitching especially apparent in sky colors.
- on adding rectification step, the stitched image is worse than without rectification step.

thanks for your insights!

fbriggs, 2016-09-12:
1. you are already using --enable_render_coloradjust, which should help. however, make sure you run your script twice. The first time it generates a file called brightnessadjust.txt, and the second time it uses it. Beyond that, there isn't much more our software can do. You may be able to improve results by changing settings on the GoPros. For example, I have had better luck in the past turning ProTune off. Part of the reason we don't use GoPros is that it is difficult to control exposure and get consistent color between them.

2. The warning you got about rectification usually indicates that it had trouble finding good keypoint matches between all images. I suggest trying again with a different scene; then you can apply the rectify.yml file you get to any other scene, as long as the cameras aren't free to wiggle.

The scene you are using looks like it is at night/very dark (mostly black in some cameras). I suggest trying again with a full-daylight scene.

knottb, 2016-09-14:
I am also having trouble with rectification using GoPros. Except I almost always get an image like this:
[image]

Looking at the keypoint matches, there are clearly errors between the images, but this is consistent across all photos I try to render. (I am starting with a fresh (blank) .yml rectification file for each attempt)

fbriggs, 2016-09-14:
Can you post an example of the keypoint matching visualization where it failed?

Short of working to improve the keypoint matching code, all you can do is try again with a different scene.

knottb, 2016-09-19:
I no longer have access to the data that produced this example, but here is an example of a keypoint visualization that produces a similar result:

[image]

fbriggs, 2016-09-19:
This image is a very difficult case for keypoint matching. You have mostly white walls, with a few corners, all of which look very similar close up. This is pretty close to the worst possible input you can give rectification. The rooftop scene you posted before should have worked better.. I'd like to see the keypoint matches on that scene.

knottb, 2016-09-19:
I can't reproduce the result from the roof image. I think I was using a different intrinsic file that I don't have anymore (poor calibration), however I still get warped rectification, just not in the same way:

[image]