facebookarchive/Surround360 #97: About the Formula i=r*sin(FOV/2-360/n)

elvindp, 2016-09-09:
I know that this formula can determine the parameters of the setup, but I still can not understand how this formula is deactivated. I would like to be very appreciate if anyone could give me some advice or instructions. Thanks!

fbriggs, 2016-09-09:
An image of the derivation is attached. I won't go through all of the details, but here's a very high level idea:

Every column of pixels in the stereo panorama that we generate is taken from the image of a virtual camera that is in between two real cameras. The virtual camera images are generated with optical flow. The virtual camera images have a lower FOV than the real cameras, because they have to overlap. The virtual camera frustum effectively rotates as we move from the left camera to the right camera. It is most rotated when it is right next to a left or right camera, so the derivation focuses on that (its the worst case). If the criteria of i = r * sin(FOV/2 - 360/n) is not met, it corresponds to trying to take a column of pixels from the virtual camera that is past its edge. The virtual camera's image is a slice of an equirectangular projection, so we work in units of angles instead of pixels, for the purpose of the derivation.

[image]

closed 2016-09-09 by fbriggs

brownliang, 2016-09-26:
Hi fbriggs : 
                     I have a little question and looking forward for your respond.
                   1. From the picture,It seems γ≠ θ，it has a little tolerance,why here they are same value?
                   2. What's the i means? i=15cm,how can I get this value,if I change another camera,such as the different camera size that is different of point grey?or the smaller device size？what's need we take care of ，if  we need to change another camera？
                   3. The f is foucus lenth? or other parameter?

brownliang, 2016-09-26:
Sorry，I make a mistake:
2. What's the r means? r=15cm,how can I get this value,if I change another camera,such as the different camera size that is different of point grey?or the smaller device size？what's need we take care of ，if we need to change another camera？

brownliang, 2016-09-26:
Hi fbriggs : 
                  I guess the r is the distance from the top of camera to the metal cycle（there are total 14 cameras） center？
                 But I am not sure calculation form the top of camera to matal cycle center is from lens top ，or the center of camera，or the CCD surface，or the back of camera？which is the right position？（Suggess the side view in 2d drawing ）
                  Another question：Formula i=r*sin(FOV/2-360/n)，the i min>3.2,if i use a camera with fov=89.7,r=15, camera=12 pieces,the result of i =3.85.
                  So it means we can reduce the camera and the volume of the surround device？

fbriggs, 2016-09-26:
r is the distance from the center of rig, to the "exit pupil" of the lenses, but that is tricky to find. It should be close enough if you use r = the distance from the center to the tip of a lens.

fbriggs, 2016-09-26:
3. f is the distance where you have zero parallax. notice we take the limit as f -> infinity, which is consistent with what you should see (zero parallax on objects at a distance of infinity).

akc2267, 2016-10-04:
Also, fov is the HORIZONTAL fov after correcting for barrel distortion.

fbriggs, 2016-10-04:
Also, fov is the HORIZONTAL fov after correcting for barrel distortion
-->
Fov horizontal AFTER correcting for barrel distortion.

BroadDong, 2016-11-16:
why the γ ( = FOV/2 + α) is the θ

BroadDong, 2016-11-16:
@fbriggs

BroadDong, 2016-11-16:
why the γ ( = FOV/2 + α) is the θ @fbriggs