facebookarchive/Surround360 #98: gopro calibration and sample files

cvaldesmol, 2016-09-10:
If someone is working with gopro can you please share the raw files and setup?
If someone has taken videos of the chessboard for calibration / or pictures can you share them (gopro 4 or with gopro 3 is fine)? I can run the calibration part and post the results here. I am really interested on seeing this work for gopro but my cameras will take a while to arrive.
My guess is that a correct calibration would be better if you set it up on video mode with the same setup you want to use to capture video and record the chessboard in many positions. I have done it for a different camera in another project and recording a video and then getting the frames from the video worked for me. I recorded the chessboard with 3 cameras / same model and found that results were a bit different so at that time I did an average on the results to represent a correct calibration

fbriggs, 2016-09-12:
There is at least one other person trying with GoPros here:
https://github.com/facebook/Surround360/issues/94

knottb, 2016-09-19:
It seems that with GoPro Hero 4 cameras, the specs vary enough from camera to camera that the stitch quality can diminish significantly. This conclusion comes from using a rig with 14 cameras around vs. a rig with one camera taking all of the pictures (in succession) in a still scene. The stitches have been much better for the single camera shots, but only when calibration is performed for the specific camera in use.

It seems the TestRenderStereoPanorama.cpp file could be edited pretty easily to take multiple intrinsic calibration files.

cvaldesmol, 2016-09-19:
knottb,
I was expecting those results since when I did calibration with a different camera brand in the past I ended up using an average. I found out that even with the same firmware and settings there are differences. Using manual settings help - nothing should be in auto.
Are you able to share your calibration videos/pictures and testing images so I can try on my end? 
Have you tried with 10 or 12 cameras? Thanks

knottb, 2016-09-19:
I am unable to share any images. I have only tried with 14 so far, but I intend to use fewer once everything is working properly with 14.

fbriggs, 2016-09-19:
If you take checkerboard images with several GoPros, then put all of the different images together in one folder and pass that to the intrinsic calibration tools, it will give you intrinsics that are probably better than if you just use one camera, then try to apply its intrinsics to all of them. Unfortunately this is an issue with GoPros. I have seen the exact same thing where rotating a single GoPro works much better.

ssaroha, 2016-09-27:
echo the comments around difference in calibration for different gopros, I ended up putting images from different cameras into one folder as well.

closed 2016-10-13 by fbriggs