facebookarchive/Surround360 #99: quesition about fisheye image angle and radius

guihui123, 2016-09-12:
It’s said that the fisheye camera Fujinon FE185C086HA-1 have a static angle 185 .
But after i use function estimateOpticalCenterFromDiffusedImage get cam15’s radius is 810,the text CALIBRATION.md said that 781 pixels and 185 degrees are the correct values for the Fujinon lens on the CMOSIS sensor in our reference design.
Is the test image use  other fisheye camera whose angle larger than FE185C086HA-1?

fbriggs, 2016-09-12:
Good question- the optical centering estimation code we provide outputs both an optical center coordinate, and a radius. However, we suggest that you only use the center cooridnate, and stick with 781 pixels and 185 degrees. These are the values from the lens maker spec sheet. The radius that you get from running the optical centering tool may vary with different amounts of illumination and/or threshold.

closed 2016-09-12 by fbriggs

guihui123, 2016-09-13:
Thank you for your reply. So you mean image taken by Fujinon FE185C086HA-1 don’t have a static angle 185. And we must use lens maker spec sheet to crop image to get the real 185 image.
By the way, i don’t find this lens maker spec sheet in the internet. Can you send this to me? It will be a huge help to me. This is my email     liukeran@360looker.com.