facebookarchive/Surround360 #105: CameraControl is running on fiber optic breakout box or camputer?

dangsiper, 2016-09-18:
Hi,

CameraControl is running on fiber optic breakout box or camputer?
If CameraControl is running on camputer, it communicates with cameras through USB port or optical fiber in CameraControl software's perspective?

So far we have not collected together all the hardware parts, so I can't test it by myself.
Thanks for your answer in advance.

BroadDong, 2016-09-18:
HI，I am testing it.Could I communicate with you?Please tell me the contact information.

aparrapo, 2016-09-18:
The CameraControl is running on the camputer. Even though the fiber optic cable is the one that is plugged to the computer, each camera is detected as USB 3.0, so from the software perspective it's the same as connecting all the cameras directly to the camputer.

The main reason why we use fiber optic is that the maximum cable length allowed by the [USB 3.0 spec](http://www.usb.org/developers/docs/whitepapers/USB_Superspeed_Equalizer_Design_Guidelines_2011-06-10.pdf) is 3m, after which the signal can attenuate. The breakout box is placed below the camera, and transforms the USB signal to fiber optic, which then can be extended to +50 meters.

dangsiper, 2016-09-19:
Thanks so much.
: )

closed 2016-09-19 by dangsiper