facebookarchive/Surround360 #106: http://127.0.0.1:8090 failed

BroadDong, 2016-09-18:
Now I run into a problem as shown below.Could you tell me how I should solve?
[image]

dangsiper, 2016-09-19:
Hi @BroadDong 
Would you please leave your email, then we can exchange our info more convenient.

BroadDong, 2016-09-19:
@dangsiper ，1418592293@qq.com

dangsiper, 2016-09-19:
Hi @BroadDong 
This problem may be caused by you haven't ran ffserver.
You should first run:
<code>
ffserver -f configs/ffserver.conf&
</code>
, then run the CameraControl program.

aparrapo, 2017-02-19:
@BroadDong did @dangsiper 's suggestion fix your issue?

aparrapo, 2017-04-22:
This version of the camera capture is no longer on the repo. Closing this issue :)

closed 2017-04-22 by aparrapo