facebookarchive/Surround360 #108: web服务器

BroadDong, 2016-09-20:
When I run the camera_control,there is the issue as shown below.Should I set up web server?And how?
[image]
[image]

dangsiper, 2016-09-20:
@BroadDong 
There are two way to view preview stream:

* First set up a web server which supports PHP, then configure the server's root to "Surround360\surround360_camera_ctl\source\www\camera_capture" or copy these files to the server's default root(It is maybe locate at /var/www/html/). At last, access URL http://127.0.0.1 and start preview.

* Direct access URL http://127.0.0.1:8090/live.webm

mybuddyhoward, 2016-09-22:
@BroadDong and @dangsiper , I'm stuck on the part of running the Camera Control software. It's throwing an error of an instance of "std::string". Seems you both have gotten further than I have. Would either of you be able to help me. You can email me @ howard@reverge.com.
Thanks in advance.

dangsiper, 2016-09-23:
@mybuddyhoward please refer to https://github.com/facebook/Surround360/issues/93

aparrapo, 2017-02-19:
#93 and #127 seem to solve this issue. Closing this one for now.

closed 2017-02-19 by aparrapo