facebookarchive/Surround360 #112: How do we get the vedios from cameracontrol?

DxgHandInHand, 2016-09-26:
When I ran “ffserver -f configs/ffserver.conf ” and “sudo ./CameraControl --nframes 5 --raw –nbits 8 --brightness 1 --shutter 20.0 --gain 0 --fps 30 --numcams 1 --debug”, I met the situation in the image.
Could you tell me how I get the videos in some detail?
[image]
[image]

mybuddyhoward, 2016-10-03:
Hey @DxgHandInHand 
Were you able to figure out how to get the videos?

aparrapo, 2017-04-22:
@DxgHandInHand this version of the capture software is no longer part of the repo. Can you try with the new version? Closing this issue for now, but feel free to open a different one if you still have problems.

closed 2017-04-22 by aparrapo