facebookarchive/Surround360 #113: About the Surrount360 device design question--Hardware

brownliang, 2016-09-26:
Hi fbriggs :
I have a little question and looking forward for your respond,thanks very much.

1. From the picture,It seems γ≠ θ，it has a little tolerance,why here they are same value?
2. What's the i means? r=15cm,how can I get this value,if I change another camera,such as the different camera size that is different of point grey?or the smaller device size？what's need we take care of ，if we need to change another camera？
3. The f is foucus lenth? or other parameter?

I guess the r is the distance from the top of camera to the metal cycle（there are total 14 cameras） center？
But I am not sure calculation form the top of camera to matal cycle center is from lens top ，or the center of camera，or the CCD surface，or the back of camera？which is the right position？（Suggess the side view in 2d drawing ）
Another question：Formula i=r*sin(FOV/2-360/n)，the i min>3.2,if i use a camera with fov=89.7,r=15, camera=12 pieces,the result of i =3.85.
So it means we can reduce the camera and the volume of the surround device ，without any 3D and surrout quality sacrificed？

brownliang, 2016-09-26:
[image]

brownliang, 2016-09-26:
[image]

brownliang, 2016-09-26:
[image]

brownliang, 2016-09-27:
Hi fbriggs :
                    Thanks for your quickly and effective reply~
                   1. As the Formula i=r*sin(FOV/2-360/n)，which the FOV will use one to calculate, the hor or diag？From the picture is used the Formula,the fov is 77°（Diag）；but in other issue replay,it requires the hor?      Pls refer to **picture_1,picture_2**
                    Sample: pls refer to **picture_3**,if we use hor Fov,the result will not meet the design requirement.   FOV=77°（Diag）=54.5（Hor）*1.414（1：1,picture H:V）
                     
Pictures_1:
[image]
Pictures_2:
[image]
Pictures_3:
[image]

brownliang, 2016-09-27:
By the way ,6.5cm seems to be the best simulation of the human eyes。if I use the r=21cm to calculate，the  eye simulation mode distance=4.65*2=9.3cm.there is a little more than human eyes.

brownliang, 2016-09-27:
FOV Sample: Hor VS Diag
[image]

fbriggs, 2016-09-27:
2. i is half the interpupilary distance that we can simulate. So for IPD = 6.4cm, we need i >= 3.2cm
3. f is not the focal length. It is the distance at which the left eye and the right eye see the same thing, i.e. where parallax is zero. The only value of f that is consistent with reality is f = infinity. Notice we take the limit as f approaches infinity in the derivation. It is just easier to do some of these calculations if we put in f as a variable, then make it go to infinity later in the derivation.

I guess the r is the distance from the top of camera to the metal cycle（there are total 14 cameras） center？
-->
Referring to your diagram, r is the distance from the "metal cycle center" to the "camera lens".

Formula i=r*sin(FOV/2-360/n)，the i min>3.2,if i use a camera with fov=89.7,r=15, camera=12 pieces,the result of i =3.85.
So it means we can reduce the camera and the volume of the surround device ，without any 3D and surrout quality sacrificed？
-->
That is correct.

As the Formula i=rsin(FOV/2-360/n)，which the FOV will use one to calculate, the hor or diag？From
-->
In the formula i = r * sin(FOV/2 - 360/n), FOV is the *horizontal* FOV of the side camera lens, after correcting for barrel distortion.

closed 2016-09-27 by fbriggs