facebookarchive/Surround360 #116: Bit more on lightfield/parallax formulas?

nsmulay, 2016-10-05:
Hi,
Was trying to understand the foll.  code and how some of the variables used afterwards; would appreciate your comments on the questions below:
---
// lightfield/parallax formulas
  const float v =    atanf(FLAGS_zero_parallax_dist / (FLAGS_interpupilary_dist / 2.0f));
  const float psi =    asinf(sinf(v) * (FLAGS_interpupilary_dist / 2.0f) / cameraRingRadius);
  const float vergeAtInfinitySlabDisplacement =    psi * (float(camImageWidth) / fovHorizontalRadians);
  const float theta = -M_PI / 2.0f + v + psi;
  const float zeroParallaxNovelViewShiftPixels =    float(FLAGS_eqr_width) * (theta / (2.0f * M_PI));
--
Questions:
1. Do you have any picture showing "psi" and "theta" angles? it would help!
    [Also, trying to understand how cameraRingRadius impacts psi.]
2. Can you elaborate more on the role of two horizontal shifts involved here: 'vergeAtInfinitySlabDisplacement'(applied while creating L/R chunks) & 'zeroParallaxNovelViewShiftPixels' (applied after L/R panorama) ? If vergeAtInfinitySlabDisplacement is already being used in warping x-coordinates differently for L/R chunks to ensure object at infinity would have zero parallax, why do we still need to use 'zeroParallaxNovelViewShiftPixels' on final L/R panoramas which were created by horizontally stacking the L/R chunks ?

fbriggs, 2016-10-13:
1. See https://github.com/facebook/Surround360/issues/97
2. This is pretty subtle and it has been almost a year since I did that derivation, but here's my best recollection:
The whole stereo panorama consists of "chunks" all horizontally concatenated. Each chunk corresponds to the portion of the panorama that comes from two adjacent cameras. When constructing each chunk, only images from two cameras are involved, and each output pixel in the chunk is a blend of two pixels, one from each camera. vergeAtInfinitySlabDisplacement modifies the coordinate of the pixel that is sampled from the source images, so it is used while producing chunks. In contrast, zeroParallaxNovelViewShiftPixels is applied after all of the chunks have been concatenated, and it shifts everything so parallax is zero at a distance of infinity. Trying to answer your question about why we need both of these- When constructing chunks, we do so for both eyes at the same time (i.e. we make a chunk for the left eye and a chunk for the right eye). The problem is, these chunks can't be at the same theta in the panorama and have it all work out so that we have zero parallax at infinity. In other words, if we wanted to construct the chunks and not apply the zeroParallaxNovelViewShiftPixels later, we would need to access pixels that do not come from only the 2 cameras that contribute to that chunk. It is necessary to be able to decompose the problem into chunks, where each chunk is affected by only 2 cameras.

closed 2016-10-13 by fbriggs

nsmulay, 2016-10-17:
Thanks fbriggs!
1. Still finding a bit tricky to understand as I think am mixing some of the symbols. Got the overall idea, but not clear when I try to map to original question 1 I had posted. I believe, from diagram: v = atanf( f/ (i/2)), but couldn't pinpoint the 'psi' yet. Can you comment OR encircle 'psi' in picture? That would be really helpful!
2. Awesome! Got the gist; thank you!