facebookarchive/Surround360 #124: exposure correction / blending error?

cvaldesmol, 2016-10-19:
I am running some test just with some frames and found out that in my scene the ceiling has issues (see ttached image)
[image]
.
Is this an exposure correction / blending error ?
Please confirm if the software does any type of exposure correction or blending
What can you advice to improve the output?
I am using gopro
Thanks

fbriggs, 2016-10-19:
There is blending, but if the two image colors are too different, the stitcher will transition from blending to picking one color (whichever it thinks is closer in the scene) in order to reduce ghosting. The seams you pointed out look like that is happening.

There may be some settings on the GoPro that help with this.

For the official Surround360 camera, we do very extensive color calibration to make it so the color is even across all cameras.

If you are using the python GUI (run_all.py), you can check "enable_render_coloradjust", and it will do some additional compensation.

closed 2016-10-19 by fbriggs

cvaldesmol, 2016-11-02:
You said " For the official Surround360 camera, we do very extensive color calibration to make it so the color is even across all cameras. "
How do you do it? how can I do the same for gopro?

fbriggs, 2016-11-02:
First, we disable all automatic color correction on the camera hardware. It may or may not be possible to do that with GoPros, but you should turn auto-exposure/protune off if possible.

Second, we take pictures of a Macbeth Chart (https://en.wikipedia.org/wiki/ColorChecker) using D65 light. Using this data, we build color correction matrices that transform the original images to something more consistent. There is more than just color correction matrices, but that is a good start.