facebookarchive/Surround360 #125: Make error for camera_ctl -- "Undefined reference"

Qwerex, 2016-10-24:
Trying to get Camera Control working and currently getting the following error:

CMakeFiles/CameraControl.dir/source/camera_control/CameraControl.cpp.o: In function `preview(surround360::ProducerConsumer<surround360::FramePacket, 1000>*)':
CameraControl.cpp:(.text+0x1807): undefined reference to `avcodec_parameters_from_context'
collect2: error: ld returned 1 exit status
CMakeFiles/CameraControl.dir/build.make:120: recipe for target 'bin/CameraControl' failed
make[2]: *** [bin/CameraControl] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/CameraControl.dir/all' failed
make[1]: *** [CMakeFiles/CameraControl.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Not sure if I'm missing any dependencies (Although I've been through the list a few times since I was getting other errors that seem to now be resolved), anyone else getting this issue?

I'm using Ubuntu 16.04 for this. Does anyone know how I could fix this?

IsaacZhi, 2016-11-07:
Maybe your ffmpeg has some problem.
Check your ffmpeg version, and how you install the ffmepg?  
Do you have installed dynamic lib ( the *.so file ) ?

dustinkerstein, 2017-01-09:
I'm running into this same issue on 16.04. Were you able to find a solution?

dustinkerstein, 2017-01-10:
Also have this exact same issue in a fresh install of 14.0.5 server.

dustinkerstein, 2017-01-10:
FYI, I fixed this particular error by recompiling ffmpeg using the default path rather than following  the ffmpeg compilation guide:

./configure --enable-shared --enable-libx264 --enable-nonfree --enable-gpl
sudo make install

aparrapo, 2017-02-19:
Thanks for the fix @dustinkerstein. ffmpeg has recently changed its installation steps, and this should be fixed by now. Closing this issue :)

closed 2017-02-19 by aparrapo