facebookarchive/Surround360 #126: Issue with Halide build on Linux 16.04

Qwerex, 2016-10-26:
So I'm trying to compile surround360_render with Halide using:

cmake -DCMAKE_BUILD_TYPE=Release  -DHALIDE_DIR="/path/to/Halide"

At first I was getting errors due to missing links, but I fixed that by adding: '${CMAKE_DL_LIBS} tinfo z' to TARGET_LINK_LIBRARIES in in CMakeLists.txt

Now I'm getting the following error when I try to make:

```
make[2]: *** No rule to make target 'cameraIspGenFast8.a', needed by 'bin/Raw2Rgb'.  Stop.
CMakeFiles/Makefile2:446: recipe for target 'CMakeFiles/Raw2Rgb.dir/all' failed
make[1]: *** [CMakeFiles/Raw2Rgb.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

Has anyone else come across this issue? I'm going to do it all again on 14.04 when I have time to see if that fixes anything, but otherwise I would appreciate if anyone had a fix.

bkcabral, 2016-10-26:
It's a typo on lines 113,.., 116 in the requisite CMakefile.txt.  Namely,  "cameraIspGen" should be "CameraIspGen."

Qwerex, 2016-10-26:
Found the typos in CMakeLists and it did the trick! Thanks!

closed 2016-10-26 by Qwerex