facebookarchive/Surround360 #127: Issues with live preview

emmanix2002, 2016-10-26:
We have successfully installed all required software to get this running, but we're currently having issues previewing the feed from the cameras either using the Live Feed URL ([http://127.0.0.1:8090/live.webm](http://127.0.0.1:8090/live.webm)) or the web camera control interface (as mentioned in issue #108 ).    

I already have `ffserver` running
[image]

We have the web interface ready as well
[image]

The problem is every time I click the **Start Preview** button, it just fails after a few seconds without showing the preview. I went through each of the `php` files and adjusted the path to the `CameraControl` executable.    
It creates the appropriate directory in `/media/snoraid`, but doesn't do anything else.     

**NOTE:** Each of the cameras is working fine when I view them using *flycapture* on my machine.    
Any suggestions?

emmanix2002, 2016-12-13:
We ended up switching over to camera _control_ui software. It's easier to use and more intuitive.

malcolmbrown, 2016-12-13:
Awesome we will give that a shot thank you!

emmanix2002, 2016-12-13:
I forgot to mention that camera_control_ui requires you to be running at least Ubuntu 16.04.

malcolmbrown, 2016-12-13:
Yes we are running 16.04. I have been following all your comments and progress as well.

emmanix2002, 2016-12-14:
If you're not seeing any feed while using camera_control_ui; you can try these tests:

1. Disconnect all cameras from your optical breakout box, then try connecting only one camera to your machine and try the preview. You can try this for all cameras (always having only one camera connected at a time)
2. With one camera already connected, try connecting a second and try the preview.

Let me know how that goes.

malcolmbrown, 2016-12-14:
Thank you so much for your help! Ok I am able to preview successfully if I use one camera.
I have tried recording a few times and the .bin files show 300 mb+ in size (as an example) but when I  try and render it completes in 1 second. Runtimes.txt says "total frames: 0".
I have tried this will 1 camera and multiple cameras.

emmanix2002, 2016-12-14:
Does it work when you have more than one camera connected at a time?
If it doesn't, then you'll need to check your `GPIO` connector, there're probably one or more with issues -- you can use a `multimeter` to confirm this. As soon as you fix that, you should be able to see a preview with all cameras connected.
When that is sorted, you can then move on to rendering your image/video.

malcolmbrown, 2016-12-14:
1: If I plug in one camera to the "USB box" then I am able to preview perfectly
2: If I plug 2 cameras in then I am able to see the cameras on the right but if I click preview then it shows only a static image.
3: if I plug all the cameras in then it shows all the cameras connected on the right but I cannot preview anything. Terminal readout" Warning: MODE_& not supported. Falling back to MODE_0. ok. ok. ok.

emmanix2002, 2016-12-14:
I can't be certain since the issue we faced was a little different, but I'll suggest you check the `GPIO` connectors to ensure they're soldered correctly. Once you've confirmed that, then try previewing with all cameras connected. All things going well, you should be able to take photos and videos at this point.

malcolmbrown, 2016-12-14:
Ok I will check the connectors. I am able to take videos using one camera, any more than that and it records nothing. Were you able to find any instructions for rendering?

emmanix2002, 2016-12-14:
That's cool. W.r.t rendering, it's quite straightforward. Here're the steps we follow on our end:

1. Open up Terminal on Ubuntu
2. Run `python /path/to/Surround360/surround360_render/scripts/run_all.py`
3. An UI interface should pop-up; here, you set all the parameters (including the steps to run)
4. If you're working with your own recording, you'll need to enable the following steps (`unpack`, `arrange`, `isp`, `render`); you can skip the `rectify` steps, it takes a while. You should also run the `ffmpeg` step for videos.

Reality7, 2017-01-18:
We ran into similar issues where everything worked fine with each individual camera by itself, but using multiple cameras ran into a deadlock grabbing the images (with RetrieveBuffer() never running into the configured TIMEOUT_INFINITE ...). Apparently, the issue was trying to grab the image from the wrong camera as the code currently will elect the camera with the lowest serial number to be the master - and our setup was configured with another camera as master :)

A quick fix would be to emphasize using the camera with the lowest serial number as your master, but it would be much more flexible if this was configurable either in the GUI or at least via config file

aparrapo, 2017-02-19:
@emmanix2002 , @malcolmbrown , @Reality7 : we have been fixing bugs in the camera control software, and this should be fixed by now. Closing this issue, but feel free to reopen it or start a new one if the problem persists :)

closed 2017-02-19 by aparrapo