facebookarchive/Surround360 #128: Sample Video Data issues

Qwerex, 2016-10-27:
When I'm processing the sample 1m video data - the raw images seem to be out of order or misnamed? I'm getting frames like the following:

[image]

I'm looking at the raw data unpacked from the binaries and the data once it has been arranged and the camera names are inconsistent with the names, some of them don't have a camera name and once arranged, many of the frames have an inconsistent number of images associated with them.

Could this be re-uploaded by any chance? I would love a demo of the rendering software before finding a 360 camera for the full rig.

fbriggs, 2016-10-27:
Please post a detailed description of any commands or GUI options you used so we can try to reproduce your issue.

fbriggs, 2016-10-27:
I just downloaded the dataset and rendered frame 0, and that seems to be fine (see attached image for output). You mentioned some of the frames have an inconsistent number of images; what frame numbers are you seeing that?

Are you running this on Mac OS X, Linux, or something else?

[image]

Qwerex, 2016-10-27:
Frame 0 is fine for me as well, however every other frame is like the one I posted. I'm using Ubuntu 16.04 to run this and my settings are as below:

[image]

I have the 2 .bin files and palace3_cameranames.txt in one folder and have copied the isp config files into the correct folder.

fbriggs, 2016-10-27:
If you run the 'unpack' step multiple times it will create duplicate copies of the images and get confused. Check your 'raw' folders to see if they have 17 files, or some other number. If it's not 17 you unpacked twice. I suggest deleting the destination folder and running again from a clean starting point.

Qwerex, 2016-10-28:
Just unpacked and arranged again from scratch in a new folder, am still getting weird results:

[image]

3 images showing the bottom cameras and 18 images in the raw folder for frame 1

fbriggs, 2016-11-02:
From discussion with other members of the team, one possibility is that you have an earlier version of the unpacker which had a bug which has now been fixed. Can you please try pulling the latest version of the code and recompiling? Sorry for the difficulties here...

Qwerex, 2016-11-10:
I've updated render to the current build and am now getting the following images when unpacking:

[image]

Although when they have been arranged, they do seem to all have 17 images in each folder unlike last time. So there's that.

fpgol, 2016-11-11:
@Qwerex 

I had the same issue with the current build and could resolve it by correcting the input arguments of the run_all.py script as follows:
- open /surround360_render/scripts/run_all.py
- go to line 45 and replace "--nbits 12" by "--nbits 8"
- go to line 46 and replace "--image_width 2448" by "--image_width 2448"
- save file

I don't know where 12bit and 2448px width in the script come from, but the images from the test footage definitely unpack correctly when using 8bit and 2048px width.

@fbriggs 
I encountered two compiler errors while trying to build the code from 2016_11_10 yesterday, which I could resolve by:
1. changing the namespace from ::gflags to ::google in /surround360_render/source/camera_isp/CameraIspGen.cpp (line 590).
2. installing libtinfo-dev on my machine (this must be new and is not mentioned as dependency in the readme-file).

An older build from 2016_09_28 compiled without errors on the same machine.

fbriggs, 2016-11-11:
Hey guys,
sorry for the continued issues here. The sample binary dataset that we provided is 8 bit, and the images are 2048x2048. So please change the lines in run_all.py accordingly (do not use 2448)!

The 2448 number is related to a different sensor which we tested. It seems the code for that accidentally landed.

fbriggs, 2016-11-11:
The issues with 8 vs 12 bit and 2048 vx 2448 should be fixed in a commit that will land later today. These will now be options that can be selected from the run_all.py GUI.

closed 2017-01-03 by fbriggs